bigtreetech/BIGTREETECH-TouchScreenFirmware #140: TFT 35 V2.0 icon labels gone after update

Vern-K, 2019-11-10:
Had some stopping mid print issues, so updated firmware now all the labels for the icons are gone, I did go into the folder and change what was some other characters back to english characters and that did nothing. I also was trying to add a button for pre heating nylon and that build failed, that was after first update when I went digging though to see if I could find what was wrong. Please keep in mind I'm new and use small words and crayons.
Thanks 
Vern
[image]
[BIGTREETECH-TouchScreenFirmware.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3828063/BIGTREETECH-TouchScreenFirmware.zip)

guruathwal, 2019-11-10:
you need to update the icons and fonts

thisiskeithb, 2019-11-10:
Did you update the icons and fonts by copying the TFT35 folder to the root of your SD card?

Vern-K, 2019-11-10:
so move this C:\Users\lucky\Desktop\BIGTREETECH-TouchScreenFirmware-master\BIGTREETECH-TouchScreenFirmware-master\Copy to SD Card root directory to update - Unified Menu Material theme\TFT35 
into 
C:\Users\lucky\Desktop\BIGTREETECH-TouchScreenFirmware-master\BIGTREETECH-TouchScreenFirmware-master\buildroot
??

guruathwal, 2019-11-10:
No
copy the TFT35 folder to your SD card, it should look like this
(your sd card drive letter):/TFT35

Vern-K, 2019-11-10:
Got it thanks, the preheat buttons are not on there but didn't get the nylon button, did get a PEGT button so will have to look into that more... I wasn't getting what folder I needed to put where.

Thanks

closed 2019-11-10 by Vern-K

my355vega, 2019-12-29:
I am having the same issue but im not understanding that fix, could someone explain more?