bigtreetech/BIGTREETECH-TouchScreenFirmware #147: Enable marlin mode on TFT35 V2

Zuru1, 2019-11-10:
hi,

I have been trying to enable marlin mode on my TFT35 V2 without any success.
What I'm wondering is, what is needed other than a rotary encoder connected to display?

Thx in advance

thisiskeithb, 2019-11-10:
Did you update your firmware?

Zuru1, 2019-11-10:
@thisiskeithb 
Yes, I have the latest available firmware, read the full readme and updated the icons. 
Still this does not work as you said i should do. I also tried the press and hold anywhere without the rotary encoder, no success there either.

bigtreetech, 2019-11-11:
hello, The LCD 12864 mode requires an additional SPI to receive commands and data from Marlin https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/2a3f376714c6ce5906ac06f772719d1d9bdf744b/TFT/src/User/variants.h#L151 At present, only TFT24 v1.1 and TFT35 V3.0 have SPI port for this function (on EXP1), while TFT35 v2.0 does not have this port, so this LCD12864 mode cannot be used.

Sthopeless, 2019-11-11:
@bigtreetech from what I can see on `STM32F103VCT6` (TFT35 v2.0) & `STM32F207VCT6 `(TFT35 v3.0) both MCU's have 3 SPI's ?!
https://www.st.com/en/microcontrollers-microprocessors/stm32f103vc.html
https://www.st.com/en/microcontrollers-microprocessors/stm32f207vc.html

AnHardt, 2019-11-11:
@Sthopeless 
It's not very likely there are more than a very few out there, being able to solder connectors for these, to the boards. About all of the 20 pins from EXT1 and EXT2 are required.
Buying an other board is by far less mess.

Sthopeless, 2019-11-11:
20pins lol.... @AnHardt have you noticed Marlins mode also works with only EXP3(?), and if you look even closer you will not all 10pins are being used..

Please if there are more MCU's being used for TFT35 v2.0 post them here I will gladly check the datasheet's of them as it stand now ALL TFT's from BigTreeTech are '12864 emulation' capable not all have the pinout available but TFT35 v2.0 has it all and just needs someone with enough knowledge whiling to to it.

Sthopeless, 2019-11-11:
[image]

Without counting Reset, +5V/+3.3V and Ground pins TFT35 v2.0 has 20 free pins.. When I have some more time I can also list which one of those 20 free pins are for SPI.

EXP2 in like 90% of the display's is only used for the SDCard and totally unrelated to this..

Sthopeless, 2019-11-14:
weekend

[image]

thisiskeithb, 2019-11-14:
Go on.

Sthopeless, 2019-11-14:
@thisiskeithb need to grab a board for more testing and is late, to be continue tomorrow.

Wouldn't it be possible to define any pins as SPI like we did in Marlin with the TMC2130's (mostly) with RAMPS board?

Zuru1, 2019-11-15:
@Sthopeless 
Would you mind sharing your progress/work on enabling marlin mode?

Sthopeless, 2019-11-15:
@Zuru1 I will asap I see it booting up marlin screen. My configs are messed up right now I had to stop because I managed to brick 2 tft's and couldn't find bootloader for v1.2..

 I have here a STM32F103 RCT6 developer board and many tft's I was trying to see if I can boot the firmware in it because it has all the pins available but no luck so far either :|

ipatch, 2019-11-16:
> child wondering in the woods here

what is marlin mode?

I have the tft35 v2 and not sure what this marlin mode you g{al,uy}s you are speakign of.

bigtreetech, 2019-11-16:
> @thisiskeithb need to grab a board for more testing and is late, to be continue tomorrow.
> 
> Wouldn't it be possible to define any pins as SPI like we did in Marlin with the TMC2130's (mostly) with RAMPS board?

Only hardware SPI is supported for slave now. Software SPI requires additional software protocol to parse data. I'm not sure whether the parsing speed can keep up with the speed sent by the master

hapklaar, 2019-11-16:
> > child wondering in the woods here
> 
> what is marlin mode?
> 
> I have the tft35 v2 and not sure what this marlin mode you g{al,uy}s you are speakign of.

It's the 12864 mode you see [here](https://ae01.alicdn.com/kf/Hf34f8f1552164fd3b2f909b64c76001aq/BIGTREETECH-TFT35-V3-0-Touch-Screen-Support-WIFI-12864-LCD-Mode-Panel-DIY-For-3D-Printer.jpg_q50.jpg)

Sthopeless, 2019-11-16:
@bigtreetech can you help us discovering if Software SPI is gonna work and speed up the process?
Make us a branch with 12864 emulation enabled for all TFT's where we define which pins are gonna be used for encoder and lcd spi. 

If we get it working maybe you can design a secondary board that plugs into the the TFT's and gives them a EXP3 port for easy connection ;)

Pixel-Soup, 2019-11-22:
Marlin mode used the encoder for menu navigation.  The tft35 does not have one.

remrut, 2019-11-22:
> 
> 
> Marlin mode used the encoder for menu navigation. The tft35 does not have one.

Did you read the first post on this issue?

Zuru1, 2019-11-22:
> Marlin mode used the encoder for menu navigation. The tft35 does not have one.

Did you even read the first post this issue?

guruathwal, 2019-11-23:
The full graphic lcd required SPI pins to work. The V2 does not have any free SPI pin on the available headers.

Sthopeless, 2019-11-24:
I haven't figure out how to use Software SPI or if it works but I have 18264 emulation running on stm32f103rct6 developer board (same mcu as v2.0 and v1.2) with 2.4", 2.8" and 3.5" TFT's.
I am still waiting for some more hardware to arrive from China so I can do more testing and I am also waiting for different stm32's MCU's maybe I can port it aswell to those boards.

MadMonkey0101, 2019-11-27:
Can we use the latest release with Unified mode on the TFT35 V2.0?? I know the 3.0 allows for classic marlin mode, would be great to get that on my 2.0 as well. but if not this unified seems like a good choice, i just dont wanna mess up my screen! Thanks!

guruathwal, 2019-11-27:
yes you can flash it to your TFT35 V2. you need to compile it for your hardware version.

MadMonkey0101, 2019-11-27:
ok thats something ill have to figure out then! thanks

MadMonkey0101, 2019-11-27:
im using a skr 1.3 , 2208s uart and this screen tft35 2.0 .  - came as a kit.. not sure how to compile the screen tho. setting up marlin 2.0 was enough of a hassle lol

guruathwal, 2019-11-27:
@MadMonkey0101 You can try the precompiled binary from the repository: [BIQU_TFT35_APP1_V2.0.25.bin](
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/BIQU_TFT35_APP1_V2.0.25.bin)

MadMonkey0101, 2019-11-27:
ook thank! wasnt sure which version to use but this answers that for the bin file thanks!!

lex007112, 2020-01-06:
Hello friends! Any progress on marlin? And another question, is it possible to connect an encoder to tft35 v2.0? Which and how?

alex920a, 2020-01-15:
ANy news? I'm trying to figure out how to connect correctly a filament runout and launch m600 properly

ipatch, 2020-01-23:
> yes you can flash it to your TFT35 V2. you need to compile it for your hardware version.

can you elaborate on how this could be done, please?

ipatch, 2020-01-23:
> Hello friends! Any progress on marlin? And another question, is it possible to connect an encoder to tft35 v2.0? Which and how?

@lex007112 

I believe it is possible to connect an encoder to the btt tft35v2 because there is a specific header breakout on the back of the board labeled encoder, so i don't see why that wouldn't be possible.

I'm gunna assume the encoder sends data via the SPI protocol, and i believe this firmware _btt tft35v2_ doesn't contain any logic presently to support an external encoder attached to the display.

I too fell into the "trap" of purchasing a btt tft35v2 and one of the major gripes i've come across is not being able to fill in mesh points using UBL with marlin as that step requires an encoder will to fill in the missing mesh points.

I remember somebody in this thread stating they were looking for a SPI protocol breakout to interface with and looking at the back of this display it could be quite possible that the _encoder_ pin header interfaces with SPI protocol / bus of the main processor on the board.

On another note, this display (_btt tft35v2_) sells for ~ $29 USD on amazon, and the display with the encoder wheel sells for $33 USD on amazon, both of which currently have a 10% off coupon, so as much as I'd like to keep my current display it seems reluctant to spend a significant amount of time interfacing an encoder wheel to the display when a new display can be purchased for ~ $4 USD more than the original display.  All that said, I'd like to keep my _btt tft35v2_ and interface an encoder wheel because I don't really want to purchase another display and wouldn't mind spending some time to interface an encoder wheel to the display if it's a total PITA.

guruathwal, 2020-01-24:
> > yes you can flash it to your TFT35 V2. you need to compile it for your hardware version.
> 
> can you elaborate on how this could be done, please?

@ipatch ... the question was about using unified menu update on TFT35V2.

ipatch, 2020-01-24:
@guruathwal 

I thought "Marlin mode" is the menu system that is used when a LCD panel has an encoder wheel for navigating the menu(s) on the LCD as opposed to a touch screen?

the original issue opened by @Zuru1 is asking about _enabling **marlin mode** on a btt tft35v2_ and I thought _marlin mode_ is something completely different from the _unified menu update_.

I could be completely wrong about all this, so please add clarity if I'm missing something here.

As it stands the menu system I have my btt tft35v2 looks like the following, which is the _unified menu system_, no?

##

[image]

---

The traditional LCD menu system aka marlin mode looks something the below, no?

[image]

guruathwal, 2020-01-25:
@ipatch don't get confused.
There are two menu layouts of the touchscreen firmware first is the **classic Menu** and second is the **unified menu**. in the later version some menu items/icons are re-ordered.

The **TFT24 V1.1** , **TFT35 V3.0** & **TFT28 V3.0** also have the **exp1** and **exp2** ports with which these variant can also act as the 12864 lcd to display the **traditional LCD menu system aka marlin mode**.
the original question of this issue was if the **TFT35 v2.0** can also be used as 12864 lcd like the **TFT35 V3.0**

On 27-Nov-2019 firmware version 25 was released which included a status screen and some other updates to the firmware after which @MadMonkey0101 asked a question about using the latest firmware update in which he referred to the **Unified Menu** as **Unified Mode** https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/147#issuecomment-559165626
To this I answered with this comment https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/147#issuecomment-559172472

ipatch, 2020-01-25:
@guruathwal 

yeah the topic of this issue got a little side tracked, as  the original issue opened was regarding the **BTT TFT35v2** and wether that _LCD touchscreen_ could enable marlin mode.  then from there people started bolting on miscellaneous questions in this issue that weren't specific to the original issue at hand.

as the topic states,

> **Enable marlin mode on TFT35 V2**

which marlin mode is the mode that the tft's you described above the tft's that have the encoder wheel attached to them, and can use the _marlin mode_ also know as the class marlin menu system.

@Sthopeless 

was working on getting that classic menu system _marlin mode_ working on a BTT TFT35v2 _touchscreen LCD **without** an encoder wheel as seen below in the picture.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/147#issuecomment-554113482

ipatch, 2020-02-07:
@ArtemTereshkovich 

unfortunately it's not cost effective get the native _marlin mode_ menu working with a touch screen, as a generic 12864 full graphics display can be purchased and shipped for under $15 USD it wouldn't make sense to get the marlin mode working on any of the touch screen only displays, other than doing it just for the sake of doing it.

I would love to say that it'd be easy, but it's not, and it's unfortunate, because  I think quite a bit of people that have touch screen displays big tree tech or not would at least like to have the option to use the marlin menu system. Another unfortunate oversight is that BTT doesn't make money in supporting their products per se, they make money in producing goods, ie. physical devices than supporting the devices via software updates, and i believe this is rather apparent when you see that they released the btt tft35v2 (not exactly certain when this was released) but the btt tft35v3 came out with an encoder wheel probably for the very reason this issue was opened.

At a $15 USD upgrade it is hard to warrant the man hours that it would cost to get the native marlin menu working on a touch screen. hence why btt just released an entirely new screen.

For my particular use case I purchased the $15 USD reprap 12864 clone display on amz and lo and behold it has the btt branding on it as well even though it was listed as biq or some other name, either way the display works via the EXP1 and EXP2 ports on my SKR v1.3 board in conjunction with the touchscreen, and haven't run into issues so far. so at best, all i can say is get a screen that natively supports _marlin mode_ and has a physical encoder wheel.

cheers 🍩

guruathwal, 2020-04-07:
On second thought, I think marlin mode on TFT35V2 might be possible.

Zuru1, 2020-04-07:
> On second thought, I think marlin mode on TFT35V2 might be possible.

Any ideas on how to do that?

lex007112, 2020-04-07:
> On second thought, I think marlin mode on TFT35V2 might be possible.

And how to implement it?

guruathwal, 2020-04-07:
Tested the Marlin Mode on BTT TFT35 V2 using 4 wires to connect the TFT to Exp1 and the Exp2 still connected to the original LCD. It took about 2 hours to get everything working.

https://www.youtube.com/watch?v=cSnfhwSTZYQ&feature=share

ipatch, 2020-04-07:
> Tested the Marlin Mode on BTT TFT35 V2 using 4 wires to connect the TFT to Exp1 and the Exp2 still connected to the original LCD. It took about 2 hours to get everything working.
> 
> https://www.youtube.com/watch?v=cSnfhwSTZYQ&feature=share

very well done. 👍

look forward to seeing how you emulate the encoder with the touch screen.

lex007112, 2020-04-07:
> Tested the Marlin Mode on BTT TFT35 V2 using 4 wires to connect the TFT to Exp1 and the Exp2 still connected to the original LCD. It took about 2 hours to get everything working.
> 
> https://www.youtube.com/watch?v=cSnfhwSTZYQ&feature=share

Cool!!! We are waiting for instructions!

guruathwal, 2020-04-13:
Here it is: https://youtu.be/3Xzh2TcboZ8

lex007112, 2020-04-14:
> Here it is: https://youtu.be/3Xzh2TcboZ8

Hello! Well done job, thanks for your hard work. Are there other options without soldering the wire to the 51 pin of the microcircuit? ... everything is very small there and you can easily make a mistake ...

ipatch, 2020-04-14:
> > Here it is: [youtu.be/3Xzh2TcboZ8](https://youtu.be/3Xzh2TcboZ8)
> 
> Hello! Well done job, thanks for your hard work. Are there other options without soldering the wire to the 51 pin of the microcircuit? ... everything is very small there and you can easily make a mistake ...

tape off the surrounding traces, so the only exposed trace is the one you need to solder. if you have kapton tape, it works great for these sort of applications. and if you're still uncomfortable, do a couple of searches on youtube for soldering.  there's no shortage of howto videos on youtube.

lex007112, 2020-04-14:
> tape off the surrounding traces, so the only exposed trace is the one you need to solder. if you have kapton tape, it works great for these sort of applications. and if you're still uncomfortable, [...]

Ok, thanks for the tip about Kapton Scotch tape :)

Raphael327, 2020-04-17:
Please help me, I checked the wiring and I have a problem:
Blank screen and only buttons in ST7920 mode
Re-check the wiring - ok
Make sure you have enabled REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER - OK
ç. Re-check the pin number in Marlin firmware with the schematic of your board - I don't know which one would be correct for SKR Pro

Thank you!
[image]

lex007112, 2020-04-17:
> Hello my friend. Once again I am facing the Pin from SKR Pro.
> 
> Please help me, I checked the wiring and I have a problem:
> Blank screen and only buttons in ST7920 mode
> Re-check the wiring - ok
> Make sure you have enabled REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER - OK
> 
> Can you help me?
> [image]
> [image]

Can I mixed up the wires when connected to EXP 1 port?

guruathwal, 2020-04-17:
@Raphael327 check all wires with a continuity tester or multimeter to see if wires are not broken

Raphael327, 2020-04-17:
@guruathwal Congratulations for your work! I can enter marlin mode but it does not emulate the image and does not exit. When I reset the card, it goes back to where it was in the photo. I think it could be the SKR PRO pins.
Can you help me?
[image]
[image]

guruathwal, 2020-04-17:
@Raphael327  did you checked which side is 5v Pin in your Exp1 Cable? you need to match it with your board schematic. it is not necessary that it is the red one.

Raphael327, 2020-04-17:
@guruathwal
Thank you very much I'm very happy :)

Now I need to see the buttons do not work. and the screen is big running over the buttons

Thank you very much.
[image]

Raphael327, 2020-04-19:
Consegui :). para funcionar SDcard utilize #define SDCARD_CONNECTION ONBOARD

ETE-Design, 2020-04-19:
@guruathwal That is very nice done, great Work...
I own an MKS TFT32, BBT's Firmware is already running on IT as you may know... But Will it be possible to do simular to this board? All schematic and pictures are Up now that MKS have Open Sourced their board / firmware...

Doenermaker, 2020-04-20:
Great work :)
Since the TFT35V1.2 also utilizes the STM32F103VCT6, this mod should also work on that screen, right?

bjoernwedi, 2020-04-29:
> Great work :)
> Since the TFT35V1.2 also utilizes the STM32F103VCT6, this mod should also work on that screen, right?

I would be interest oed on this also, because TFT35V1.2 is quite cheap meanwhile,

bjoernwedi, 2020-04-29:
What about hte BTT TFT35 V1.2 pins U-52,U-53,U-53,U-33 and RESET on the SD breakout. Shall be 80% of SPI2 already. Except U-51 is missing, but U-33 is there.

Zuru1, 2020-05-02:
@guruathwal  Nicely done. Will try this out as soon as I can.
One question, if I connect the encoder button to the encoder port on the TFT and still wire the two wires from that connection port to the exp1 or 2, will this work as the v3 does?

digant73, 2020-06-04:
> @guruathwal
> Thank you very much, I'm very happy :)
> It's working really well :)
> Now I can't save the error settings on the sdcard in the EEPROM. Do you know it can be?
> When I click to save configuration on the SDcard it will make the home
> Thank you very much.
> [image]

but what was the problem and the fix you made?

ETE-Design, 2020-06-04:
@guruathwal Any chance on making this working on the MKS TFT 28/32?

[7 more replies]

closed 2020-08-14 by bigtreetech