bigtreetech/BIGTREETECH-TouchScreenFirmware #151: Can not load correctly after compiled in Visual Studio code

tphan26363, 2019-11-11:
I make change to has dual extruder to the TFT-24 controller.  It compiled sucessfully is Visual studio Code.  However when upload through SD card, after initial copy of the firmware.  The TFT-24 appear unable to boot and load new code and just blackout. The  changes as :

;BIGTREE_TFT24_V1_1
[platformio]
src_dir      = TFT
boards_dir   = buildroot/boards
default_envs = BIGTREE_TFT24_V1_1  // change here

[common]
default_src_filter = +<src/*> -<src/Libraries> -<src/User/Hal/stm32f10x> -<src/User/Hal/stm32f2xx>
build_flags = -fmax-errors=5
  -g
  -ggdb
  -DUSE_STDPERIPH_DRIVER=
  -D__STATIC_INLINE=
  -ITFT/src/Libraries/cmsis/Core-CM3
  -ITFT/src/User/Fatfs
  -ITFT/src/User/Hal
  -ITFT/src/User/Menu
  -ITFT/src/User
  -ITFT/src/User/API
  -ITFT/src/User/API/UI
  -ITFT/src/User/API/Gcode
  -ITFT/src/User/API/Language
  -ITFT/src/User/API/Vfs
  -ITFT/src/User/Hal
  -ITFT/src/User/Hal/STM32_USB_HOST_Library/Core/inc
  -ITFT/src/User/Hal/STM32_USB_HOST_Library/Class/MSC/inc
  -ITFT/src/User/Hal/STM32_USB_HOST_Library/Usr/inc
  -ITFT/src/User/Hal/STM32_USB_OTG_Driver/inc
  -DSOFTWARE_VERSION=24.2 //keep this version unchanged.

It also could be because Visual Studio Code was automatically update some thing.  Please let me know if you have similar problem and how to resolved it.  The change I made to the code is so minor.  I don't think that is the reason that the firmware can not be loaded.  I also check the hardware and firmware version and they appeared to be correct.
Thanks
TP

Hukuma1, 2019-11-11:
Had this issue on the TFT35 myself. But then they pushed the last commit about an hour ago. It fixed it for me. Can you try again?

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/aa773a140a29b27de7dbabebd0a06bb38336fa5e is the commit that fixed it for me.

tphan26363, 2019-11-11:
It may fix for TFT-35 But not for TFT-24.  I just download a couple minutes ago and it is no joy.
TP

thisiskeithb, 2019-11-11:
Can any other TFT24 users who compiled their firmware confirm this issue?

Don't forget to update PIO and clean your libraries/delete you `.PIO` folder before exporting.

tphan26363, 2019-11-11:
Did all the clean up the build folder and Update PIO.  Still no joy.

tphan26363, 2019-11-11:
Also there are many warning as below when build.   I don't think it is serious issue though.

FT\src\User\Menu\Mode.c: In function 'menuMode':
TFT\src\User\Menu\Mode.c:93:11: warning: unused variable 'nowEncoder' [-Wunused-variable]
   int16_t nowEncoder = encoderPosition = 0;
           ^~~~~~~~~~
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Printing.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\ProbeOffset.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\ScreenSettings.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\SendGcode.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Settings.o
TFT\src\User\Menu\Print.c: In function 'gocdeListDraw':
TFT\src\User\Menu\Print.c:98:10: warning: implicit declaration of function 'bmpDecode' [-Wimplicit-function-declaration]
       if(bmpDecode(strcat(gnew,"_70.bmp"), ICON_ADDR(ICON_PREVIEW+i)))
          ^~~~~~~~~
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\Speed.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\UnifiedHeat.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\UnifiedMove.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\leveling.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\Menu\menu.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\delay.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\main.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\my_misc.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\src\src\User\os_timer.o
Archiving .pio\build\BIGTREE_TFT24_V1_1\lib118\libCommon.a
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_adc.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_adc_ex.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_can.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_cec.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_cortex.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_crc.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_dac.o
In file included from C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal_def.h:46:0,
                 from C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal_rcc.h:45,
                 from C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal_conf.h:234,
                 from C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal.h:46,
                 from C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_can.c:161:
C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_can.c: In function 'HAL_CAN_GetTxTimestamp':
C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\CMSIS\Device\ST\STM32F1xx\Include/stm32f1xx.h:210:32: warning: implicit declaration of function '__CLZ' [-Wimplicit-function-declaration]
 #define POSITION_VAL(VAL)     (__CLZ(__RBIT(VAL)))
                                ^
C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_can.c:1097:23: note: in expansion of macro 'POSITION_VAL'
     transmitmailbox = POSITION_VAL(TxMailbox);
                       ^~~~~~~~~~~~
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_dac_ex.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_dma.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_eth.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_flash.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_flash_ex.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_gpio.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_gpio_ex.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_hcd.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_i2c.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_i2s.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_irda.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_iwdg.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_mmc.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_nand.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_nor.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_pccard.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_pcd.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_pcd_ex.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_pwr.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_rcc.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_rcc_ex.o
Compiling .pio\build\BIGTREE_TFT24_V1_1\FrameworkHALDriver\Src\stm32f1xx_hal_rtc.o
[... cut ...]
C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_pwr.c:114:92: note: in expansion of macro 'POSITION_VAL'
 #define CSR_EWUP_BB(VAL)         ((uint32_t)(PERIPH_BB_BASE + (PWR_CSR_OFFSET_BB * 32U) + (POSITION_VAL(VAL) * 4U)))
                                                                                            ^~~~~~~~~~~~
C:\users\tphan\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_pwr.c:403:22: note: in expansion of macro 'CSR_EWUP_BB'
   *(__IO uint32_t *) CSR_EWUP_BB(WakeUpPinx) = (uint32_t)ENABLE;

thisiskeithb, 2019-11-11:
Those warnings were from a previous version of firmware. Download a fresh copy and try compiling again.

thisiskeithb, 2019-11-11:
@tphan26363: Does this binary work? [BIGTREE_TFT24_V1.1.24.2.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3833171/BIGTREE_TFT24_V1.1.24.2.bin.zip)

tphan26363, 2019-11-11:
The original binary work but the one I try to recompile does not work since I want to change to dual extruder

thisiskeithb, 2019-11-11:
@tphan26363: Can you try the one I uploaded? I compiled it today.

tphan26363, 2019-11-11:
Ha. I download the zip it does not work so I used GitHub Desktop,  It works.  guess they did not update the zip file. Thanks every body
TP

closed 2019-11-11 by tphan26363

tphan26363, 2019-11-11:
@thisiskeithb It does work.
TP