bigtreetech/BIGTREETECH-TouchScreenFirmware #152: icon description?

Aleks-MO, 2019-11-11:
There are so many icons, can someone describe what they are responsible for? Can make a map of icons?

thisiskeithb, 2019-11-11:
They are named based on their function. What info are you looking for specifically?

Aleks-MO, 2019-11-11:
Lightbulb icons? When printing, is the MORE icon at the bottom right?

thisiskeithb, 2019-11-11:
The light bulb icons named `RGB_*` are for controlling RGB LEDs and the `More` screen is for controlling your fan, LED color, extruder, filament sensor, etc.

Improved documentation is in the works, but keep in mind that this firmware is being actively developed by BigTreeTech and the community, so these features are subject to change.

Aleks-MO, 2019-11-11:
Ok/

thisiskeithb, 2019-11-12:
[Here's an older guide](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/blob/f8e52a5aafb071d30a86764b5d5fd72eda0a5430/TFT35%20screen%20customize.pdf) in the TFT35 V1.2 repo, but many icons and screens have been updated since that was originally posted.

Aleks-MO, 2019-11-12:
This is how to collect new icons, you can explain to children what is responsible for what. Colored icons are more informative, red “heat”, Blue “movements” green “action” 
[image]

guruathwal, 2019-11-12:
> Lightbulb icons? When printing, is the MORE icon at the bottom right?

When printing the third icon in second row is Percentage.bmp o to open the speed and flow setting.
Tune.bmp makes more sense.

kkaleta, 2019-11-18:
> This is how to collect new icons, you can explain to children what is responsible for what. Colored icons are more informative, red “heat”, Blue “movements” green “action”

These icons will not match the latest firmware and I do not plan to update them. But I'm working on something new, less childish but still colorful.
[image]

thisiskeithb, 2019-11-18:
> These icons will not match the latest firmware and I do not plan to update them.

Any plans to release the source files so we can try to bring it up to date? It’s a nice icon pack, but wouldn’t be usable for anyone other than those with the exact firmware version you’re running.

Aleks-MO, 2019-11-18:
I'm sorry. Everything turns into a mess .. (It seems time to change to TFT MKS.

closed 2019-11-18 by Aleks-MO