bigtreetech/BIGTREETECH-TouchScreenFirmware #157: [Suggestion] Global status icons/info

Hukuma1, 2019-11-12:
[image]

We have a bunch of unused space at the top because it's reserved for menu titles (where it says "Ready") and an occasional error message printed in the middle. If you did not want to revamp that at all, could we at least add icons/information to the top right corner? This way you could always have info of your noz/bed temperature no matter what menu option you are in. I think eventually you could get rid of the titles/errors altogether and replace them with icons and other global information.

What do you all think?

Sthopeless, 2019-11-12:
I agree @Hukuma1 something like the new prusa mini tft (looks fabulous btw :P )

[image]

Hukuma1, 2019-11-12:
That does look great! Well they got the right idea. Let's do it as well. Although lol @ 1/3 of their display gone to logo + telling you what screen you're in. D:

Sthopeless, 2019-11-12:
@Hukuma1 (and others :P) check this out https://www.youtube.com/watch?v=ipulB4_Xdm8 the screen also has menus marlin alike, looks very cool..

hapklaar, 2019-11-12:
Did you see this? https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/153

Hukuma1, 2019-11-12:
Yep, looks great! The idea here is to put some of the same information into the top corner and have it be there globally for all menu selections. Even simple text only to start off would be a big improvement towards knowing the status of your printer.

arminth, 2019-11-15:
Too bad it doesn't compile for 2.4" version:
[image]
When setting to 3.5, it compiles.

Hukuma1, 2019-11-17:
[image]

I think this global status bar being persistent in all of the menu options (except in the new main screen that @guruathwal made) will greatly improve printer awareness.

Sthopeless, 2019-11-17:
uhhh! @Hukuma1 is your branch available for testing/download?

Hukuma1, 2019-11-17:
No, I'm unfortunately not a coder. :/ Just posting crude mockups from Photoshop. Hoping some talented coders here can implement some of the ideas though.

Hukuma1, 2019-11-17:
[image]

Made a less crude mockup. Here is my proposed printing screen.

Sorry, @guruathwal, the STOP button was burning my retinas. ;)

The clock icon next to nozzle/bed/fan icons is the time remaining. Time elapsed is in the top right corner.

And even a 'reali life' example for @Aleks-MO :)
[image]

Sthopeless, 2019-11-18:
I liked more with the temps on the right side of the screen. Maybe even have changing colors when powered on(? :P ).

The preview of the file is cool and stuff but sometimes I have many files that are the same with different settings, I rather much more have a readable file name then see what the file i sliced looks like.

Hukuma1, 2019-11-18:
I tried the colors changing with values at first, but it started looking like an Xmas tree. :( As far as file name, you will see full file name at top left corner? (e.g. CE3_benchy) Even more room since global icons are not at the top either. Or what did you mean?

Aleks-MO, 2019-11-18:
Dear colleagues, please complete the final version. The mess in the solutions. I am already confused which BIN file for which icons.
@Hukuma1    
I really want to see this during my lifetime ....)))))

guruathwal, 2019-11-19:
@Hukuma1 you can fit only 3 lines in an icon with text height of 24px for TFT35.

Hukuma1, 2019-11-19:
@guruathwal No way around that? Possible shrinking the text height to accommodate a 4th line? If not, then I guess worst case is we move the time remaining to the top right and have it switch between elapsed and remaining every 5 seconds? Trying to avoid people touching the device in case they're afraid of bumping the printer.

guruathwal, 2019-11-19:
@Hukuma1 youa have great editing skills. this might be helpful https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/153#issuecomment-555188661

guruathwal, 2019-11-19:
Will start working on the files menu by the end of this week.

Hukuma1, 2019-11-19:
@guruathwal https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/153#issuecomment-555188661 Saw this, can't wait for this to get merged! Will be the first thing to get people to use touch screen over native marlin I think.

The 3 lines you mentioned. Does that mean it would look something like this if we put 3 lines into icon space? Meaning no room for icon to the left? :(

[image]

guruathwal, 2019-11-19:
> @guruathwal [#153 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/153#issuecomment-555188661) Saw this, can't wait for this to get merged! Will be the first thing to get people to use touch screen over native marlin I think.
> 
> The 3 lines you mentioned. Does that mean it would look something like this if we put 3 lines into icon space? Meaning no room for icon to the left? :(
> 
> [image]

@Hukuma1  yes will be like it.

Hukuma1, 2019-11-19:
@guruathwal What if you remove center alignment on the text. Is there enough room to push a BMP icon to the left of the text then?

Otherwise another solution (maybe better) would be to use fixed-width icons as a font. Take a look here: https://github.com/ryanoasis/nerd-fonts

Has ability to add any/all icons (Material, FontAwesome, etc) to any font. Already has patched font for what we use (Ubuntu Mono: https://github.com/ryanoasis/nerd-fonts/tree/master/patched-fonts/UbuntuMono) in TFT35. This will also enable us to change colors of the icons dynamically if needed because it's simply text.

Btw the new home screen is awesome. Great work! And thank you for fixing up the STOP button. It all looks great now. Can someone change the translation of "Percentage" to maybe "Adjust" under Print screen?

guruathwal, 2019-11-20:
@Hukuma1 We will have Icons embeded in fonts for the top status bar and some new live icons to replace some repetitive icons like the preheat and custom gcode menu.

In printing menu we can have one wide status box just like the one in status screen, it can fit a preview image, 3 lines of 24 px icon with 7 characters.
If we omit the preview icon we can have full live info in that box.

Hukuma1, 2019-12-31:
[image]

Looks like upcoming commits (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/272) by @chzj333 add these suggestions in. Thank you! Would still love a possible icon solution for the status info, but this is already great.

closed 2020-03-19 by Hukuma1