bigtreetech/BIGTREETECH-TouchScreenFirmware #158: TFT3.5 V3 bootloader

artistan82, 2019-11-13:
Was playing around with SWD, creating a customized firmware and accidentally overwrote the bootloader.

Not all update needs to be performed via SWD.

Does anybody have a dump of the bootloader for TFT3.5 V3?

Baltazart, 2019-11-13:
sorry for my English
First load the boot, then update for TFT35 / v3.0 it should revive LCD. there is a valid drop box link.
hoping to help you.

https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/20

thisiskeithb, 2019-11-13:
You’ll see my comments in the issue, but I can confirm it works 🙂

guruathwal, 2019-11-13:
> sorry for my English
> First load the boot ST-LINK, then update for TFT35 / v3.0 it should revive LCD. there is a valid drop box link.
> hoping to help you....
> 
> [bigtreetech/BIGTREETECH-TFT35-V3.0#20](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/20)

👆This should be added to the repository.

thisiskeithb, 2019-11-13:
I’ve asked BigTreeTech to open source it like the TFT35 V2, but was told it wouldn’t be for now.

guruathwal, 2019-11-13:
@thisiskeithb can you point to TFT35 V2 bootloader source, i cant find it? all I found was the BIN file.

Baltazart, 2019-11-13:
> @thisiskeithb peut -être allumé la source du chargeur d'amorçage TFT35 V2, je ne trouve pas? tout ce que j'ai trouvé, c'est le fichier BIN.

there is this boot, but apparently load by usb, see in picture.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT35_V2%20Bootloader%20fix

Baltazart, 2019-11-13:
there is this boot, but apparently load by usb, see in picture.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT35_V2%20Bootloader%20fix

guruathwal, 2019-11-13:
> there is this boot, but apparently load by usb, see in picture.
>
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT35_V2%20Bootloader%20fix

👍  Thanks @Baltazart ... but I already know that. I pushed the PR for this one. @thisiskeithb is saying that the TFT35 V2 Bootloader is open source. I was asking about the location of the source code.

Baltazart, 2019-11-13:
> 👍 Thanks @Baltazart ... but I already know that. I pushed the PR for this one. @thisiskeithb is saying that the TFT35 V2 Bootloader is open source. I was asking about the location of the source code.

Sorry!

thisiskeithb, 2019-11-13:
I misspoke. They didn’t open source it, just provided a copy.

guruathwal, 2019-11-13:
@thisiskeithb ok. atleast the Binary for TFT35 V3 should be added to this repository.

artistan82, 2019-11-13:
Confirmed working! Thank you so much.
I agree that they should have this open sourced.

closed 2019-11-13 by artistan82

Baltazart, 2019-11-13:
> Confirmed working! Thank you so much.
> I agree that they should have this open sourced.

Happy for you ... lol