bigtreetech/BIGTREETECH-TouchScreenFirmware #159: Full flash rom Bigtreetech TFT35 V2.0

marinercinco, 2019-11-13:
I have erased the complete bigtreetech tft35 v2.0 chip to be able to upgrade to tft35 v2.0.23 firmware.
Once updated, it does not recognize bigtreetec skr pro v1.1 "No printer attached"
You can supply me with the full flash rom to restore the communication of the RS232 port.
Thank you

thisiskeithb, 2019-11-13:
Did you try loading the bootloader from [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT35_V2%20Bootloader%20fix)? You should be able to update once that’s loaded.

marinercinco, 2019-11-13:
That has allowed me to update firmware, but with the accidental deletion of the chip there is no communication through the RS232 port.
I need to reload the complete rom of the chip to reestablish communication through the RS232 port.
Thank you

HappyCodingRobot, 2019-11-14:
The bootloader together with the firmware are the complete ROM. The configurations are rewritten by the firmware. I intentionally erased the flash before writing the bootloader.
What interface do you use to delete and write? ST-link, BMP, jlink .. ?
Can you check the adresses (0x08000000 for BL and 0x08003000 for FW) ?

marinercinco, 2019-11-15:
> 
> 
> The bootloader together with the firmware are the complete ROM. The configurations are rewritten by the firmware. I intentionally erased the flash before writing the bootloader.
> What interface do you use to delete and write? ST-link, BMP, jlink .. ?

The total erase of the chip was with ST-LINK V2 of BAYTE and the prog. STM32 ST-LINK Utility.
Later I use the STM32CubeProgrammer.
Regardless of the bootloader, program or interface I use, I cannot communicate via the RS232 port.

ipatch, 2019-11-15:
yeah if your using the tft35 v2 then some batches of those boards had issues with their bootloader and loading new firmware.  i purchased a board from amz about a week ago and was able to compile the FW source without a problem and update the firmware on my lcd.

If your experiencing communication issues between your skr board and your lcd, then that may be a separate issue.

- are you able to communicate with your board using a different lcd (if you have one available) ?
- have you tried loading the precompiled marlin FW on your skr and the precompiled FW on your lcd provided by big tree tech?

personally if it were me, i would try using the precompiled versions both of the marlin fw for the skr board, and the precompiled fw for the lcd, and if your experiencing issues still, then it's more than likely a completely separate issue, but if you've been compiling marlin and lcd fw from source it is very easy to misconfigure things to where the lcd won't com with the board, i've experienced this first hand. 🙋‍♂️

guruathwal, 2019-11-15:
> The total erase of the chip was with ST-LINK V2 of BAYTE and the prog. STM32 ST-LINK Utility.
> Later I use the STM32CubeProgrammer.
> Regardless of the bootloader, program or interface I use, I cannot [...]

you should check your TFT35 Serial by connecting its rs232 port to your computer via any USB to TTL converter any check both  Baudrates 115200 and  250000. you should see "M105" received every few second if the TFT if working properly. be sure to cross-connect the Rx & Tx pin of TFT to TTL converter.

marinercinco, 2019-11-17:
Hello,
Following its instructions, the RS232 port is dead. It does not respond with anything, it does not communicate ...
I have ordered a TFT35 V3.0.
Thank you for everyone's attention and help.
regards

guruathwal, 2019-11-17:
Did you check to make sure the rx and tx pin are connected corectly?
Rx of usb TTL to Tx of TFT
Tx of usb TTL to Rx of TFT

marinercinco, 2019-11-17:
> 
> 
> Did you check to make sure the rx and tx pin are connected corectly?
> Rx of usb TTL to Tx of TFT
> Tx of usb TTL to Rx of TFT

Hello,
I have carried out the checks thoroughly and with different USB-TTL modules. I can only connect to the TFT through the SWD port.
In the tests I lost the touch of the TFT with the BIQU_TFT35_APP1_V2.0.24.2.bin firmware and recovered it by returning to the BIQU_TFT35_APP1_V2.0.24.bin firmware.
It would be possible to obtain the initial BIQU_TFT35_APP1_V2.0.20.bin firmware to try to recover the communication through the RS232 port. Does anyone have it?
I will be and I am very grateful for the help.
regards

CharlesSee, 2019-11-18:
Hi. I'm also in the process of figuring out the TFT35 v2. 

I got the same issue with the touch screen not working after updating the bootloader and reflashing the firmware using atom etc. But I just reverted back to the stock BIGTREE_TFT35_APP1_V2.0.24.2.bin  and the touchscreen is now back working. 

I am using an arduino Uno with the gnd/reset pins jumpered together, and RX to RX and TX to TX, GND to GND and 5V to 5V. It works like a USB to TTL bi-passing the chip/script. 

I am using the FlyMcu.exe software, and couldnt get the TFT to respond. 

I was using tweezers to jumper the pins on the TFT and pressing reset, getting the white screen, and THEN clicking the FlyMcu.exe "Start ISP(P) button. Usually the TFT didnt respond NO matter what BAUD rate I chose in FlyMcu. and if it did respond if it liked the baud rate it was garbage characters.

My trick is this : 

Select any Baud rate in FlyMcu, mine is 57600. 
Have the TFT connected, but NOT in the white screen.
Click on the FlyMcu "Start ISP(P)" first, it starts timing out with no response, and THEN quickly, using tweezers to connect the TFT pins and pressing reset and thus whiting out the screen. 
Then the TFT seems to auto-negotiate the baud rate and the bootloader flashing now works.

Maybe this might help..

CharlesSee, 2019-11-18:
Sorry I don't have any original.bin file. But also when I changed the "settings" "baud rate" on the tft via its menu from "115200" to "250000" to match the rate you have in your printers firmware, the "no printer attached" message immediately goes away and the tft is able to control the printer..

ipatch, 2019-11-18:
> you should check your TFT35 Serial by connecting its rs232 port to your computer via any USB to TTL converter any check both Baudrates 115200 and 250000. you should see "M105" received every few [...]

is there a way to prevent the lcd from constantly sending the m105 cmds to the board, i like using the gcode console to send gcode to the board for diagnostic purposes, so when i send my cmd, the ouput disappears because the lcd is constantly sending m105 commands.

marinercinco, 2019-11-22:
> 
> 
> Did you check to make sure the rx and tx pin are connected corectly?
> Rx of usb TTL to Tx of TFT
> Tx of usb TTL to Rx of TFT

Hi,
I still have the problem of the "no printer attached" principle.
For the SWD port I have rewritten bootloader and updated firmware without problem.
Could someone provide me with the electrical scheme to verify the board and the components associated with the RS232 port.
Thanks in advance
Cheers

guruathwal, 2019-11-23:
@marinercinco here is the manual for tft35v2 https://github.com/bigtreetech/BIGTREETECH-TFT35-V2.0/blob/master/TFT3.5-V2.0%20manual.pdf

marinercinco, 2019-11-23:
Hi,
First of all apologize for my English, I use the Google translator.
The schematic does not come in the manual and I cannot check wire and components between RS232 and the MCU STM32F103VTC6.
I am very grateful for your attention.
Cheers

guruathwal, 2019-11-23:
@marinercinco Bigtreetech has not release the PCB schematics for the TFT.
you have to refer to the pinout of the TFT35V2 and [STM32F103VCT6 datasheet](https://www.st.com/resource/en/datasheet/stm32f103vc.pdf) and and check the PA9 & PA10 pins 

for your reference:
[image]
[image]

CharlesSee, 2019-11-23:
Thanks for the reply
It was the baud rate.
You can close this issue.
thanks, Charles Seebacher

From: marinercinco
Sent: Saturday, 23 November 2019 1:44 AM
To: bigtreetech/BIGTREETECH-TouchScreenFirmware
Cc: CharlesSee; Comment
Subject: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] Full flash romBigtreetech TFT35 V2.0 (#159)

Did you check to make sure the rx and tx pin are connected corectly?
Rx of usb TTL to Tx of TFT
Tx of usb TTL to Rx of TFT
Hi,
I still have the problem of the "no printer attached" principle.
For the SWD port I have rewritten bootloader and updated firmware without problem.
Could someone provide me with the electrical scheme to verify the board and the components associated with the RS232 port.
Thanks in advance
Cheers

CharlesSee, 2019-11-23:
Thanks for the reply..
regards, Charles Seebacher

From: guruathwal
Sent: Saturday, 23 November 2019 10:21 AM
To: bigtreetech/BIGTREETECH-TouchScreenFirmware
Cc: CharlesSee; Comment
Subject: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] Full flash romBigtreetech TFT35 V2.0 (#159)

@marinercinco here is the manual for tft35v2 https://github.com/bigtreetech/BIGTREETECH-TFT35-V2.0/blob/master/TFT3.5-V2.0%20manual.pdf

marinercinco, 2019-11-23:
> 
> 
> @marinercinco Bigtreetech has not release the PCB schematics for the TFT.
> you have to refer to the pinout of the TFT35V2 and [STM32F103VCT6 datasheet](https://www.st.com/resource/en/datasheet/stm32f103vc.pdf) and and check the PA9 & PA10 pins
> 
> for your reference:
> [image]
> [image]

Thanks for the help. For me it has been very instructive and a starting point to solve the problem.
Cheers

closed 2019-11-23 by marinercinco

my355vega, 2019-12-29:
I am having an issue with the flashing on my tft35 v2.0 i have literally tried every download and changed the file name to every option and cant get it right. I was able to flash with 2 dif bin files and it worked and both had dif screen layout except with both the icons are half blank and the wording is in like a dot matrix. idk what im doing wrong here or what i need to do it right? very new to this trying to get the screen to work on my skr mini e3 1.2 and not having luck. anyone have any advice or solutions?

guruathwal, 2019-12-29:
@my355vega  did you updated the icons and fonts?

CharlesSee, 2019-12-29:
Yes thanks, you can close this. I have updated the firmware and fonts etc..

guruathwal, 2019-12-29:
@CharlesSee  good to know that..😊.. but I was asking to @my355vega as he just commented with the problem..

CharlesSee, 2019-12-29:
Hi my355vega,

I assume you have the bootloader updated so that the firmware can be uploaded, that was my first problem. It was never updating the firmware from the one it was shipped with even though it looked like it was.

Then using the github bigtreetech touchscreenFirmware I just copied from the directory "Copy to SD Card root directory to update - Unified Menu Material theme' \TFT35\*.* directory to my SD card root as \TFT35\*.*  and my compiled  BIQU_TFT35_APP1_Vxxxx.bin file to the root also. (I compiled my settings using ATOM. 
Firstly, if BIQU_TFT35_APP1_Vxxxx.bin exists, (xxx being the version which changes, like BIQU_TFT35_APP1_V2.0.24.2.bin), then the firmware updates first.
Then if \TFT35\*.* exists, the icons etc update, and it creates the file TFT35.CUR.
You then need to delete the files \TFT35\*.* or BIQU*.bin otherwise each time you turn on the display it will repeat the update.

Are you saying that the default files dont work, or have you modified them ?

my355vega, 2019-12-29:
Thank you for getting back to me and with more information. Reviewing your
reply CharlesSee I might have over looked a step. Threw sd card I was able
to flash the firmware TFT35_v2.0.25 following with the folder TFT and a
reset. Which has given me most the icons back threw the menu's and the font
language is back to normal. I think I could still be missing a step. I have
only been transferring files over by sd card. Haven't done anything threw
atom yet.

CharlesSee, 2019-12-29:
Hi.
So, on the tft, under menu settings, info, it says the firmware version is 2.0.25 ?
Compiling is only required if you want to change things like the number of tools, or extruders, the preheat temperatures, printer size, custom buttons etc in BIGTREE-TouchscreenFirmware-Master\TFT\src\User\Confguration.h.
After compiling BIGTREE-TouchscreenFirmware-Master\platformio.ini with ATOM, I then copy the new BIGTREE-TouchscreenFirmware-Master\.pio\build\BIGTREE_TFT35_V2_0\BIQU_TFT35_APP1_V2.0.25.bin file over to the sd card root directory. It should work ok.
I haven't bothered with editing icons yet, so I cant help you there...

my355vega, 2019-12-29:
That is correct on the firmware version. looking at it now I think I downloaded an older bin file for the 2.0.24 if that would have made a difference. this is where im at right now.
[image]
[image]
[image]

CharlesSee, 2019-12-30:
I'm using the SKR Pro board which might be a little different.

The "no printer attached" message for me was the baud rate, on the TFT35 menu, under settings, UART speed. make sure it matches your compiled SKR mini Marlin Code you are using on your board under configuration.h 👍 

BAUDRATE 250000 or 115200 etc.
SERIAL_PORT -1  // (the usb cable connection on the board)
SERIAL_PORT_2 1 // for TFT (-1 usb port, 1 TFT, 6 WIFI module) must be set.

Marlin only supports 2 serial ports I believe.

So if you want to use the WIFI module, you can plug it into the TFT35 and it works, but it also has to be setup which is quite tedious. You cannot have TFT and WIFI on the main board at the same time if you want to use octoprint connected to the usb port on the board. But you could have it if you don't use octoprint, like SERIAL_PORT -1, SERIAL_PORT_2 1.

And of course the cable, that should be plugged in the right way around.

hope that makes sense.

my355vega, 2019-12-30:
Awesome thank you for all your help figuring this out. Ill be jumping into the marlin next to get it to connect properly with the board. I thought it was a firmware issue on the screen and now that its fixed and knowing its programming threw marlin ill begin my new adventurers there.

CharlesSee, 2019-12-30:
also if you need help with the bigtree esp8266 1mb wifi card setup if you have it, just let me know. 

this video is great : https://www.youtube.com/watch?v=pJGBRriNc9I&t=544s
Its not necessary to use the voltage regulator he mentions, I just used an Arduino Uno by itself, jumper cables and breadboard. Hopefully you will not get any errors when loading the ESP and SPIFFS. If you do it will depend on what the errors are..

I had an issue as I was uploading the esp sketch, it always said 'card not found' and timed out. If you reconnect the ESP8266 3v line cable 'AS' you are uploading and it WILL renegotiate the baud rate and will works. the error is missleading. must be as and when you are uploading , eg , you have already pressed the upload button, not before.

When plugging into the TFT35 v2, plug in facing antenna outward, just works, no cable required. my baud rate is 250000 on the SKR board side.

Cable used to plug ESP8266 into Arduino for uploading ESP firmware and SPIFFS: 
TX -> RX
RX -> TX
GND -> GND
3V -> 3V  (attach after pressing upload button not before)
EN -> EN

It should just work, temp readings show its working via the web browser ip address.
TFT35 v2 will show its ip address and confirm when you start the TFT35.
no config necessary on the TFT35 v2. but a litte bit slow. 

The reason why I'm still using octoprint, is because according to the above video the file upload is slow whereas octoprint is fast and I use the octoprint webcam to monitor my printer. But I have the esp8266 plugged into the TFT35 v2 just because I have it already and it was only a few dollars, and in case I have octoprint issues. And I originally thought I could have the 3 serial devices plugged into the mainboard, but you cant, so my SKR Pro mainboard has the USB for octoprint, the TFT35v2 and the TFT35 v2 has the esp8266 WIFI card plugged into it.

my355vega, 2019-12-30:
Ill was actually going to order the wifi adapter but at the time of search
I noticed people where talking about 3 different ones so I decided to wait
till I got everything set up and learned before jumping into it. but after
reading threw your message it doesnt seam to be as hard as I thought it
would be. Ill definitely give you a shout as soon as I reach that point,
thank you so much. out of curiosity I decided to look at the pin out from
my tft35 and the pin out on the board for the tft and discovered i have the
plug flip around on the board so I reversed it and now the screen works
with the board and all is communicating properly. that was exciting to see.
thank again for all your help and guidance Ill be sure to reach out with
any questions down the road.

CharlesSee, 2019-12-31:
sorry, I removed the cable pin layout from the previous message as it was wrong.  just use what he says in the video. TX to TX etc.