bigtreetech/BIGTREETECH-TouchScreenFirmware #164: Toggle between temperature units

ipatch, 2019-11-15:
Recent owner of a tft35 v2, and was messing around setting temperatures last night and noticed and could not figure out how to toggle between **celsius** and **fahrenheit** and more comfortable in understanding **fahrenheit** as opposed **celsius**.

is there way to set the default units for temperature to **fahrenheit** or even better be able to toggle between the two units?

##

[image]

guruathwal, 2019-11-16:
it is not implemented yet.

dineshvyas, 2019-11-19:
You can convert temperature to Kelvin without any problem but Fahrenheit is not compatible with the metric system. However, it is possible.

gordo3di, 2020-09-05:
@ipatch  Can you please close this ticket. Thanks

ipatch, 2020-09-05:
why does it need to be closed?

oldman4U, 2020-09-07:
@ ipatch.

Of course, you do not have to close the ticket if you do not want. But soon this ticket is one year old and it seems that there is no interest to add Fahrenheit. When you look at the code, everything is defined in ºC.

On the other hand, closing old and unused tickets helps to keep this ticket system useful. So it would be great if you could close this ticket.

Thank you

oldman4U, 2020-09-07:
Something i really would like to know.

When you buy filament, do you get the recommended printing temperature printed on the spool also in Fahrenheit? I have never seen such a spool, so i would like to know if they print different labels for different areas of this world.

I mean i understand you, because we got used to inches when it comes to computer displays and TV screens and the industry failed completely trying to change this (at least here in Europe).

closed 2020-11-13 by stale[bot]