bigtreetech/BIGTREETECH-TouchScreenFirmware #169: Tft35 v3.0.24.2 update problem

josephsaathoff, 2019-11-17:
It seems that the icons and fonts are not uploading to the screen. Have white icons and unable to read the writing.

ipatch, 2019-11-19:
I've noticed something similar when updating the FW for my tft35v2 this morning, all the icons on the default "home" screen have a certain fuzzyness to them, but if click in _(press menu)_ from the "home" screen all the icons from within the _menu_ button appear to render as they should the below gif is an example of what I'm describing above.

##

<div align="center">

[image]

</div>

guruathwal, 2019-11-20:
You need to update your icons. There are new icons for the status screen which needs to be updated on the TFT.

ipatch, 2019-11-20:
~~do i need to copy both folders to the root of the SD card along with the bin or just one folder?~~

```
Copy to SD Card root directory to update - Unified Menu Material theme~~
Copy to SD Card root directory to update~~
```

... should have read the instructions more clearly 🤦‍♀️

##

**leaving this here for future me** _the contents for the sd card should look like the following if lcd = tft35v2

[image]

guruathwal, 2019-11-20:
you need to rename the `TFT35.CUR` folder to `TFT35`

ipatch, 2019-11-20:
yeah that was the contents of the dir after i flashed.

guruathwal, 2019-11-20:
👍🏼👍🏼 you can close the issue now if the problem is solved.. :)

ipatch, 2019-11-20:
unfortunately it's not my issue to close, i was just commenting on it.

guruathwal, 2019-11-21:
@ipatch Well then we have to wait for @josephsaathoff to confirm..

josephsaathoff, 2019-11-21:
My issue was as card too big and was not formatted to fat32

closed 2019-11-21 by josephsaathoff