bigtreetech/BIGTREETECH-TouchScreenFirmware #180: Please add Z axis !

Aleks-MO, 2019-11-24:
Please add Z axis. Printing on the screen lacks Z axis readings. It is very helpful sometimes when you know which layer is printed.
[image]

bigtreetech, 2019-11-25:
ok, This will be added, Thank you for your advice.

kkaleta, 2019-12-02:
And (maybe) we don't need this "T" before time? 
[image]

Aleks-MO, 2019-12-02:
@kkaleta  Class!     Many wanted to know the print layer. 
I do not understand what the letter "T" is?

kkaleta, 2019-12-02:
Actually this one look even better. With GRAY font and time and Z on top.
[image]

Aleks-MO, 2019-12-02:
I don’t know, I need to work a week .... the first option is all the data in one revenge and immediately perceived as a single whole ..) In general, the habit of the MKS screen remained.

Aleks-MO, 2019-12-02:
Can you put the firmware + icons, where is the Lyme firmware? The more people try, the more they think the best.

Aleks-MO, 2019-12-02:
@kkaleta Oh, oh, oh, oh, I just saw you - Relay-controlled-by-phone-call-Advanced! How does it work? Doesn't hang? I got a "monster" ..)))))    http://avrproject.ru/forum/13-123-1

closed 2020-07-04 by bigtreetech