bigtreetech/BIGTREETECH-TouchScreenFirmware #181: tft 3.5 v1.2 does not print from sd

antonello69, 2019-11-24:
Hi everyone, I have a motherboard mks gen 1.4 and a tft 3.5 v1.2 I updated marlin to bugfix version 1.1 I updated the tft firmware to 1.2.24 but it doesn't print anymore from sd someone can help me thanks

lijinlin19970813, 2019-11-25:
Did your printer display "Ready", if the display is ready, check if the SD card slot is good.

lijinlin19970813, 2019-11-25:
> Hi everyone, I have a motherboard mks gen 1.4 and a tft 3.5 v1.2 I updated marlin to bugfix version 1.1 I updated the tft firmware to 1.2.24 but it doesn't print anymore from sd someone can help me thanks

Did your printer display "Ready", if the display is ready, check if the SD card slot is good.

antonello69, 2019-11-25:
thanks for the reply the display tells me no printer attached but everything works fan, extruder, movements and also browses the sd but when I click on a gcode file it does not print

guruathwal, 2019-11-27:
Did you checked the Serial Speed? It should match the serial speed set in Marlin firmware

antonello69, 2019-11-27:
yes I have checked they are at 250000 then it works almost all the movements of the axes work the fan works the extruder heats up and even the dish heats up even the sd I can browse the files but when I try to print a gcode file the print does not start

antonello69, 2019-11-27:
[image]

one thing that is wrong according to me is this writing no printer attacched!

lijinlin19970813, 2019-11-28:
> [image]
> 
> one thing that is wrong according to me is this writing no printer attacched!

Check the serial port and baud rate of your marlin firmware.

guruathwal, 2019-11-28:
@antonello69 the TFT will treat the printer as connected if it receives temperature reading from the printer.
if you are able to actually move all axis of the printer then go to the Gcode menu and see if the TFT is sending the "M105" command and receiving the temperature reading from the printer.

antonello69, 2019-11-28:
> > [image]
> > one thing that is wrong according to me is this writing no printer attacched!
> 
> Check the serial port and baud rate of your marlin firmware.

#define SERIAL_PORT 0
#define BAUDRATE 250000
these are the settings I have on marlin but if it was wrong I could not even move the axes or heat the flat extruder and turn on the fan

guruathwal, 2019-11-28:
@antonello69 can you check the gcode menu on TFTand see if it is sending M105 commands and receiving replies from the printer?

antonello69, 2019-11-28:
@guruathwal  I understand I send a m105 from gcode of the tft and then I should get the temperature where?
however I sent the command m105 but it doesn't seem to me that I get temperatures anywhere

[image]
[image]

guruathwal, 2019-11-28:
you do not need to type M105.
just go to Gcode menu and press send button. it will open terminal screen.
there you will see 'send M105' displaying every few seconds and reply from the printer if both are able to communicate.

antonello69, 2019-11-28:
@guruathwal if I press the send button nothing happens but if I send the command m105 it opens (I think it's the terminal but all black I get nothing) I attach the photo
[image]

antonello69, 2019-11-28:
anyway thanks to all for the interest and sorry english i write with the translator

carl1961, 2019-11-28:
What printer board you using?

guruathwal, 2019-11-29:
@antonello69 it looks like you are using old version of the firmware.. in latest version the send and back buttons in gcode menu are larger..

antonello69, 2019-11-29:
I'm using as a motherboard a gen 1.4 mks and tft 35 v1.2.24 as firmware and marlin bugfix 1.1 but I tried them also the bugfix 2.0 I can't print from tft if I connect the laptop to the printer with repeater host everything works to wonder

guruathwal, 2019-11-29:
you need to update your TFT firmware. `v1.2.24` does not show data sent or received in terminal. use the latest version here : 
[BIQU_TFT35_V1.2.25.bin](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/BIQU_TFT35_V1.2.25.bin)

the terminal screen should look like this if it does not receive any response from the printer

[image]

antonello69, 2019-11-29:
@guruathwal I put the firmware 1.2.25 now I see send m105 but no answer and the print is still wrong I can not understand what does not print
[image]

guruathwal, 2019-11-29:
@antonello69 ok.. now we have some thing....
check the continuity of the wire connecting Tx of main board to Rx of the TFT. it looks like one wire in the connection cable is broken.

antonello69, 2019-11-29:
OK I try

antonello69, 2019-11-29:
@guruathwal I tried the cable looks good maybe some settings in the marlin? what motherboard do you have and what version of marlin?

guruathwal, 2019-11-30:
@antonello69  I have SKR V1.3 running 2.0.x
did you tried reducing the serial speed to 115200 on both your board and TFT display?

antonello69, 2019-11-30:
yes I also tried to 115200 today I put an arduino mega on tx rx and when I click on the gcode to print nothing happens while when I click on other I receive the relative gcode commands

antonello69, 2019-11-30:
the beauty is that before with marlin 1.1.8 and a firmware of the tft it seems to me 1.2.7 everything worked then I updated marlin and tft and it doesn't print anymore from tft. of course from repater host everything works

guruathwal, 2019-11-30:
> the beauty is that before with marlin 1.1.8 and a firmware of the tft it seems to me 1.2.7 everything worked then I updated marlin and tft and it doesn't print anymore from tft. of course from repater host everything works

did you disconnected the usb before using the TFT? TFT will not work properly if the main board is connected to USB on PC.

guruathwal, 2019-11-30:
also try rolling back to the 1.1.8 marlin firmware

antonello69, 2019-11-30:
es of course I disconnected USB I tried everything is more than a month anyway I also got in touch with bigtreetech maybe they can solve the sebra problem as if the buttons to print gcode files are disabled it seems that even the button for the fan half way does not work

guruathwal, 2019-11-30:
@antonello69  your TFT is not able to receive data from the board. untill it receives any data it will not start printing.
can you post the photo your wiring of the TFT and the board about which pin in connected to which?

antonello69, 2019-11-30:
I don't know if it makes the idea
[image]

antonello69, 2019-11-30:
live
[image]
[image]

guruathwal, 2019-11-30:
looks good...
so you confirm that the Red and Orange wire are not broken.
if you have a usb serial converter then can you try connnecting the TFT directly to PC with the serial converter to check if it is passing the data?

antonello69, 2019-11-30:
red and orange I confirm they are good tested with testers I then took the tx and rx from the tft and I sent them on the arduino mega and I saw that it always passes m105 and when I click on the various icons fan movement etc they pass the relative gcode but when I click on the gcode file icon to print nothing passes as if it were disabled

carl1961, 2019-11-30:
try swapping the TX and RX     RX goes to TX and TX to RX     I had to do
that on a S6 serial header

antonello69, 2019-12-01:
I tried to swap rx to tx and tx to rx but nothing in this way doesn't even work the fan movement etc.

antonello69, 2019-12-01:
i have installed the unified firmware 1.2.25 still nothing i attach the photo
[image]

carl1961, 2019-12-01:
ok the look at your marlin code you may need the serial 2 set to 1
this is what mine is set too in Configuration.h
#define SERIAL_PORT -1

#define SERIAL_PORT_2 1

antonello69, 2019-12-01:
I have a mks gen 1.4 it has only one serial I still tried but it doesn't go

guruathwal, 2019-12-01:
> looks good...
> so you confirm that the Red and Orange wire are not broken.
> if you have a usb serial converter then can you try connnecting the TFT directly to PC with the serial converter to check if it is passing the data?

did you tried this?

do this and try sending data text from pc while gcode terminal is open in the TFT..

antonello69, 2019-12-01:
I'll try

antonello69, 2019-12-02:
I did the tests with usb serial converter cable and it seems that the motherboard does not send anything so the tft does not receive any response to the command m105

antonello69, 2019-12-07:
@guruathwal when the display sends m105 which code it receives from the motherboard thanks?

antonello69, 2019-12-09:
inally I managed to test the motherboard serial seems to work well because sending M105 responds with ok T: 34.55 /0.00 B: 22.62 /0.00 @: 0 B @: 0 now you need to understand if the tft is not good this type of response that of you

antonello69, 2019-12-09:
Finally I managed to test the motherboard serial seems to work well because sending M105 responds with ok T: 34.55 /0.00 B: 22.62 /0.00 @: 0 B @: 0 now you need to understand if the tft is not good this type of response that of you?

guruathwal, 2019-12-09:
> Finally I managed to test the motherboard serial seems to work well because sending M105 responds with ok T: 34.55 /0.00 B: 22.62 /0.00 @: 0 B @: 0 now you need to understand if the tft is not good this type of response that of you?

Yes this is the response you should get from the main board. you need to connect the TFT to USB serial with correct serial speed and send this line: `ok T: 34.55 /0.00 B: 22.62 /0.00 @: 0 B @: 0` and check the TFT if it shows the temperature on the status screen.

antonello69, 2019-12-11:
nothing to do the tft doesn't really want to receive anything I don't know if it's a firmware issue or it's broken but I don't know what else to do

closed 2020-08-14 by bigtreetech