bigtreetech/BIGTREETECH-TouchScreenFirmware #182: Wich IO-Port is for the filament runout sensor?

oOJoshOo, 2019-11-24:


HappyCodingRobot, 2019-11-24:
in variants.h :
V3:
FIL_RUNOUT_PIN PA15

V2:
FIL_RUNOUT_PIN PD11 -> TZ2 connector

oOJoshOo, 2019-11-24:
Thank you very much now it works

HappyCodingRobot, 2019-11-25:
@oOJoshOo Please close the issue, if this is solved

guruathwal, 2019-11-27:
@oOJoshOo  please close the issue.

closed 2019-11-27 by oOJoshOo

arut16, 2020-01-25:
> Thank you very much now it works

Hi, 
Can you please explain how did you do to make it working ? In my case, nothing happend if filament is runout !

oOJoshOo, 2020-01-27:
@arut16 have you changed the the settings in your firmware?

arut16, 2020-01-27:
> @arut16 have you changed the the settings in your firmware?

In Marlin on my SKR 1.3 ? I changed all settings according to the BTT manual

oOJoshOo, 2020-01-27:
Ok than i dont know, ive made my own sensor with an old endstopp and on the tft 2.0, also i used Visual Studio, because Platform ide and Atom dosen't work anytime