bigtreetech/BIGTREETECH-TouchScreenFirmware #185: Unknown Command popups (Prusa Slicer)

dexter323i, 2019-11-24:
The display is TFT35 V3.0, latest FW. I use Prusa Slicer. I printed from Display SD port, and during the print, it kept popping up messages telling echo: Unknown command. Two examples:
"M73 P58 R3" ok
"M73 Q30 S5" ok

[image]
[image]

bigtreetech, 2019-11-25:
hi, according to http://marlinfw.org/docs/gcode/M073.html, M73 is a set current print progress percentage for LCD(LCD12864/LCD2004 etc...) manually, it's useless for  touch screen,You can enable `LCD_SET_PROGRESS_MANUALLY` in Marlin's `Configuration_adv.h` to support this command, or disable this commad in your slicer

guruathwal, 2019-11-25:
M73 is used to set the progress bar value like this `M73 P12` to set at 12%. There are extra variables in your g code file for M73 command which are not compatible with marin firmware.
You need to check your slicer setting and gcode file.

dexter323i, 2019-11-25:
Okay, I accept M73 P__ is the standard format, and Q R S are Prusa specific, not standard parameters. But can you modify the FW, not popping up this window? Hide the error, that's all. This could solve the issue.

I guess Marlin 2.0 FW does not now this advanced feature also, but the factory Ender 3 display not pops up errors. Just steps over it.

guruathwal, 2019-11-25:
@dexter323i  you can disable the M73 command in prusa slicer
make sure `Supports remaining times` is unchecked in prosa slicer settings as below:

[image]

dexter323i, 2019-11-26:
@guruathwal thank you so much! This checkbox solved the issue!

closed 2019-11-26 by dexter323i

thisiskeithb, 2019-11-28:
`M73 R` support was added to Marlin 15 days ago: https://github.com/MarlinFirmware/Marlin/pull/15549

Marlin’s documentation needs a refresh, so it’s not listed.

MBlackmagic, 2024-01-01:
I have written a python script and a windows droplet for this purpose. You can drop your gcode file onto the M73exDroplett and it removes all M73.
https://github.com/MBlackmagic/M73ex/tree/main