bigtreetech/BIGTREETECH-TouchScreenFirmware #186: Bed temperature not reading correct.

carl1961, 2019-11-25:
just installed BIGTREE_TFT35_V3.0.25.bin, display TFT 3.5 3.0 (first version that came out). not a problem with printing, just a Let You know.  bounces from 60960 to -099
[image]
[image]

carl1961, 2019-11-25:
Might be mad at what I'm printing LOL

guruathwal, 2019-11-26:
Is it displayed corectly on the printer lcd?

carl1961, 2019-11-26:
Glad you ask, Yes it shows correctly, and when your in print screen it shows correctly, but soon as it homes and starts printing  it shows the random numbers like in first picture. the homing is issued from the print gcode file, it may be something in my startup. I will check
[image]
[image]

carl1961, 2019-11-26:
My startup g code, shows nothing temp related
G28 ; home all axes
G92 E0.0
G1 Z5;
G1 X10 Y10 Z0 F6000;
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0

guruathwal, 2019-11-26:
Did you set the bed temperature at 60 both times?

carl1961, 2019-11-26:
No the first was preheat, the 60 came from the Gcode

carl1961, 2019-11-26:
Simplify3D

G90
M82
M106 S0
M140 S60
M190 S60
M104 S200 T0
M109 S200 T0
G28 ; home all axes
G92 E0.0
G1 Z5;
G1 X10 Y10 Z0 F6000;
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0
; process Process1
; layer 1, Z = 0.294
T0
G92 E0.0000
G1 E-4.0000 F1800
G4 P1

guruathwal, 2019-11-26:
ok.. it looks like the `/` in bed temperature text is replaced by `9`.

carl1961, 2019-11-26:
tried with Cura, and same thing. If no body else has this issue it most likely on my end.

guruathwal, 2019-11-26:
I checked and found that the `/` between the current temp and target temp is drawn only once when the printing page loaded.
its been overdrawn by other value.
Try opening the percentage menu or heat menu when this happens then go back and check if it fixes.

carl1961, 2019-11-26:
Yes guruathwal that fixed it for about 20 or 30 seconds and you have to do it again then again it goes with the odd numbers

guruathwal, 2019-11-26:
ok.. will fix it in next PR.

carl1961, 2019-11-26:
No hurry, printing works fine.

carl1961, 2019-11-26:
Oh while I am thinking about it, what is that yellow dot at top right that pulses on and off like every 10 or so seconds.

guruathwal, 2019-11-26:
thats busy icon, when the printer sends busy signal.

bigtreetech, 2019-11-27:
The bug of temperature display is fixed here.https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/6ae27e59a5e489cd79b78577d9bae566a060ca3e 
The precompiled file will be updated later

closed 2019-12-02 by carl1961