bigtreetech/BIGTREETECH-TouchScreenFirmware #190: Custom GCode: string multiple commands?

Hukuma1, 2019-11-25:
[image]

Toying around with the new custom menu and I want to make sure it's working properly. It looks like this can string multiple codes, just the popup (M117) seems to show everything at once (and no popup at end). Any ideas?

Using latest v25 firmware.

#define CUSTOM_0_LABEL "ABL 1click"
#define CUSTOM_0_GCODE "M117 ABL 1-click start!\nM190 S60\nG28\nG29\nM500\nM501\nM190 S0\nM117 ABL 1-click done!\n"

Should be:
display start message
preheat bed to 60
home
ABL
save ABL mesh to EEPROM
load saved settings
cooldown bed
display end message

Oh btw, any M117 command in start Gcode in slicer also does not seem to be placed into the status notifications. Each M117 ends up in a popup like above.

guruathwal, 2019-11-26:
> Code should do the following:
> 
> display start message
> preheat bed to 60
> home
> ABL
> save ABL mesh to EEPROM
> load saved settings
> cooldown bed
> display end message

The TFT sends the custom GCode to the printe all at once as it is declared in one string.
the printer will perform the steps in sequence of the gcode specified.

> Oh btw, any M117 command in start Gcode in slicer also does not seem to be placed into the status notifications. Each M117 ends up in a popup like above.

if you are on status screen then M117 message will show up in infobox, or if you are on any other menu then it will show a pop up so that you do not miss the last message from the printer.

Hukuma1, 2019-11-26:
So no way to execute m117 all the way at the end of the sequence? Not a dealbreaker, just want a way to let me know what code it’s on if in the middle of the sequence.

guruathwal, 2019-11-26:
@Hukuma1 try `M118 E1` like `M118 E1 Start 1Click ABL! ` this will tell printer to echo text to the host.

Hukuma1, 2019-11-26:
@guruathwal I don't see the text. I'm guessing the echo command only shows up in some terminal screen? Sorry, pretty new to Marlin and 3D printers still. :P

guruathwal, 2019-11-26:
@Hukuma1 
`M117` is used to display Message on printer LCD.
`M118 E1` is used to send Message from printer to host like PC or the TFT controller. M118 message will not be shown on printer's LCD.

Hukuma1, 2019-11-26:
Gotcha. Okay, closing this non-issue. Glad the actual Gcode sequencing works as expected. Will leave above example for anyone looking to do similar tasks. Thanks!

Hukuma1, 2019-11-27:
Just found an issue after all.

If I remove the M117 from the start it begins the whole sequence, but does not wait for the bed to heat up to 60. If I leave the M117 command in the front (popup shows up), then it does it in proper sequence. Bed must heat up to X temp, and then it starts Home, etc, etc.

guruathwal, 2019-11-27:
can you try the same gcode sequence through PC via USB?

Hukuma1, 2019-11-28:
Using Pronterface and USB, M117 commands do not show a popup on the printer. Not even on the new status home screen.

M118 E1 commands just echo back in Pronterface and that's it.

Sending the first command (M190 S60) does not update the bed temp until it fully reaches 60 degrees. So it will sit on the screen saying 30/0 for example and then once it reaches target temp it will just jump to 60/60.

closed 2020-08-14 by bigtreetech