bigtreetech/BIGTREETECH-TouchScreenFirmware #192: TFT35 V1.2.25 update trashed

FlailAway, 2019-11-26:
Hi, this is the first time I have used this and factory firmware worked OK, but touch was bad. On the back, the board has
  BIGTREETECH TFT35 V1.2
 
I downloaded the TFT35 V1.2.25.bin and the font and bmp files and copied to the SD. Plugged in SD and booted and it ran through the Font, Icon (66) and Firmware updates. 0% to 100%

It rebooted and showed a new Orange+White Bigtreetech logo and after a few seconds I get a screen with

white square -- white square -- white square -- white square 
Blue line full width with fuzzy characters in centre
white square -- blue box with moving line --  white square.

Touch does not work at all.

How can I get this to work?

guruathwal, 2019-11-26:
you need to update the icons and fonts.
re-download the fonts and icons from this repository. old icons and fonts will not work.
make sure to use icons form Unified menu folder.

FlailAway, 2019-11-26:
Thanks but I said I had done that. I tried both Unified and the Ubuntu fonts. I have since tried rolling back to v1.2.24.2 but same mess on the screen.

[image]

FlailAway, 2019-11-26:
And, yes, I did clean off the SD before adding the earlier update to it.
It showed 
  icon-pic-15     15/(66)

and went all the way through to 66, so it is clearly adding something in, just not the correct stuff.

guruathwal, 2019-11-26:
try reformatting your sd card to fat32

FlailAway, 2019-11-26:
Thanks, but I had done that too. But it is now working. 

I worked my way back to v1.2.20 and it is working but the Touch is terrible still. I have an MKS-TFT32 and the touch is excellent on that. 

Will be using this sad waste of $28 to help fill my trash bin, the Touch is totally unusable like this.

guruathwal, 2019-11-26:
[image]

moag1000, 2019-12-07:
I have exactly the same as in the picture and tried the same fixes (format sd in fat32 and start over, swap sd card, ...) any ideas?

carl1961, 2019-12-07:
Copy TFT folder to sdcard, if you have a folder already on sdcard delete it first . After reboot, verify the TFT folder renamed to TFT.CUR

imonsei, 2019-12-16:
I'm having the same problems.
Comparing the bmp images from the old version to this one, I see that the old bmp files are 32bit encoded, and the newer ones are 24bit.
Luckily I made a backup of my old firmware and fonts, so I'm at least able to use my printer.
@FlailAway let me know if you want a copy of my working 1.2.11 bin files.

moag1000, 2019-12-17:
> 
> 
> I'm having the same problems.
> It seems that if you just put the BMP and FONT folders in the root directory, the firmware still loads them, but into the wrong memory banks. you HAVE to place them in TFT35\bmp and TFT35\font before it loads correctly.
> Maybe it would be a good idea to perform a file placement sanity check in the code?

That's exactly what I did wrong, too. T.b.h. I did not go through all folders to read readme files, so I ran into the same issue. After finding the readme file I feel a bit dumb, but at least I'm not alone.

It would still be great to fix this behaviour, because this could also happen with leftover stuff on an SD card in some cases.

Mindinmatrix, 2019-12-19:
Its all fixed, but now i have no touch screen....urgh

guruathwal, 2019-12-19:
@Mindinmatrix you can reset the screen by creating a text file in SD card with the name: `reset.txt` and then put the sd card in the TFT and press reset button.

Mindinmatrix, 2019-12-20:
@guruathwal that did not work for me. I dont know what i'm doing wrong.

guruathwal, 2019-12-20:
@Mindinmatrix are you able to update the fonts and Icons?

Mindinmatrix, 2019-12-20:
@guruathwal , I was using the 1.2.25 file(Because I have V1.2 not V2) and it did not work. So i compiled my own copy using Atom (+platform IO) added storePara(); in boot.c . I then used the reset.txt on SD card as you suggested and VOILA all fixed.
This thread helped me understand the reset file:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/204 

Nevertheless  thank you so much you helped alot!

guruathwal, 2019-12-20:
@Mindinmatrix  thats great..!

closed 2020-08-17 by bigtreetech