bigtreetech/BIGTREETECH-TouchScreenFirmware #193: Bumpy benchy

jsbowman6, 2019-11-26:
[image]
The orange benchy was printed with the stock Ender LCD and SKR 1.3 board.  The green one was from the BTT TFT 3.5 V3.0 and the SKR 1.3.  
BTT had me install the latest TFT software and use the Marlin 2.0 Bugfix for SKR 1.3.  The result was a much less bumpy benchy.  Seems like it may be a communication error.  The TFT also will not go into simulator mode and BTT is sending me another TFT to try, so this could contribute to my benchy issue.  The https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/firmware/Marlin-bugfix-2.0.x/Marlin/Configuration.h software has a few issues, like not turning on my hotend fan, so I got to find out it's purpose after cleaning several jams.

guruathwal, 2019-11-27:
Try printing ths same gcode without TFT through PC or SD card on the board. This will pinpoint the issue. If printed well, its TFT problem, if printed with same issue then Marlin issue

jsbowman6, 2019-11-27:
With TFT removed the benchy printed is the orange one and printed fine.  Anytime and anyway I added the TFT into the mix, the benchy would produce bumps.  BTT had me update the SKR 1.3 firmware and the TFT's firmware at the same time, so we can't determine which fixed it.  Having done that though, the TFT still will not go into Simulator mode.

hapklaar, 2019-11-27:
Not sure if this is related, but I'm experiencing communication drops when connected through TFT's ESP port. Communication issues might also cause bad print results.

I only receive output from TFT after sending data to the port. Then it stops again, visible in temperature measurement output. Currently I have Octoprint connected to that port as this is the only way to get M117 output on TFT, but it's not working as Octoprint is depending on these temp readings.

Then again I'm still not certain on how everything should be connected ideally, to make use of all features...

jsbowman6, 2019-12-11:
This issue went away with the loading of Marlin 2.0 SKR 1.3 and the latest TFT firmware.  BTT had me do both so not sure which one fixed it.

guruathwal, 2019-12-18:
@jsbowman6 please close the issue

Piscanc, 2019-12-27:
to narrow it down a bit, i am experiencing same (similar) issue with e3_mini (marlin2.0.1) and 35v2.0 (latest btt fw).
if i print of internal sd and start from tft, i get nice prints.ž
if i print of LCD sd, so started from lcd i get same bumps. pauses correlate with blinking status (i presume its communication status) pixels in the lcd corner.

and another thing, if i start print from tft, during printing i can no longer interact with lcd, i get "busy processing..." mesage, witch is expected to a point, but not for entire print.
or is that caused by marlin, so it insures that nothing can interfere with printing process... avoiding artifacts?

jsbowman6, 2019-12-27:
"if i start print from tft, during printing i can no longer interact with lcd"  Although I no longer get the bumps, I cannot access the LCD when printing from the BTT TFT 3.5, V3.0 on the SKR 1.3 board.  I considering disconnecting the ESP 1 and 2 ribbon cables and connecting the ESP 1 back to the original Ender 3 LCD.  It worked well that way when I was troubleshooting.

Piscanc, 2019-12-27:
to correct me previusly... 
tft, meant original ender 3 screen,
lcd, meant tft35v2.0 
my mistake... just to be on same page :)

CRCinAU, 2020-02-29:
Ok - so you've hit on a problem that goes much deeper than you realise.

I initially reported this against Marlin - as you can monitor the planner buffer with ADVANCED_OK set. What happens is that when the TFT sends a M114 to get the current position, this calls `planner.synchronize()` within Marlin. This means that the command halts until the planner buffer is empty. Once the buffer is empty, the position is reported and things continue on again.

The problem is, when the planner buffer gets empty, printing stops for a fraction of a second - and this produces your lump.

As such, we really need to stop the TFT sending M114 - as this will cause micro-pauses in the print to fix this ASAP. As to what the long term fix is, I'm not sure....

hfog, 2020-02-29:
@CRCinAU Yeah, I think you're on to something here. I'm not sure that M114 is the only thing going on here but it definitely is contributing. I've been experimenting with printing from Octoprint through the TFT's UART to the printer and I've noticed these pauses too and when printing very fast, Marlin can completely freeze up. This conversation has led me to experiment with changing the TFT's screen off of the main status screen to some other menu screen that doesn't show stats such as coordinates or temps. In that mode it shouldn't be polling for positions or temps. That seems to improve things significantly.
More testing is necessary but so far it's promising.

denis-itskovich, 2020-03-08:
I think it can be closed, since #16894 fixes it

closed 2020-03-20 by bigtreetech