bigtreetech/BIGTREETECH-TouchScreenFirmware #194: Screen shifts to right in TFT (conventional) mode.

dexter323i, 2019-11-26:
I have a TFT35 V3.0, latest FW. Main board is SKR Mini E3 V1.2.

Sometimes I ran into a strange issue when I used the display in convetional TFT mode (when rotary encoder is used). If I rotated the knob too quickly or I don't know how, but suddenly the screen slided a bit into the right, and the left part is filled with the right part, a bit shifted. Hard to describe... I successfully recorded a video where I could reproduce the problem:
https://youtu.be/x85rvSQO4Z8
(Not listed video)

Hukuma1, 2019-11-27:
Can confirm I have been getting this same bug since even firmware TFT v23.

brushby, 2020-01-15:
i get exactly the same problem

davehardy20, 2020-01-15:
I get this issue also on a regular basis

billydragon, 2020-01-15:
The same happen for me. Also some time it look like picture. Printer still running.

[image]

dexter323i, 2020-01-30:
Okay. It seems I have found a solution. (Maybe workaround?) Changed the baud rate to 250000, and solved the issue for me. With this setting, only some caharcters goes crazy, but after more rotation on the knob, it repairs itsfelf.
My settings in Marlin:
- platformio.ini ln 45: monitor_speed = 250000 (In [env] section)
- platformio.ini ln 336: monitor_speed = 250000 (in [env:STM32F103RC_bigtree_512K] section)
- Configuration.h ln 106: #define SERIAL_PORT 2
- Configuration.h ln 115: #define SERIAL_PORT_2 -1
- Configuration.h ln 126: #define BAUDRATE 250000

Then in the settins menu of the TFT screen in Touch mode, and done!

closed 2020-01-30 by dexter323i