bigtreetech/BIGTREETECH-TouchScreenFirmware #195: Gcode Command

Jingsun-Young, 2019-11-27:
First I print with end gcode that work well in marlin mode. but when print in touch mode, it stuck & shown error notification
[image]
[image]

After I skip some gcode line, it work well until shutdown. 
[image]
[image]

I'm not sure this is the issue or I miss understand for something. Please help.

bigtreetech, 2019-11-27:
On the first issue: Can you send me your gcode so that I can reproduce the question? 
about the second issue:  When you enable the AutoPower in Feature menu(Settings->Feature or Printing->More->Feature),  After printing, The screen will wait until the temperature of the nozzle is lower than 50 ℃ (for safety) and automatically add "M81" command to shut down. You can choose force to shut down immediately or cancel the  shutdown.

Jingsun-Young, 2019-11-27:
Thanks for your advice. I'd send you gcode file attached. The second was not issue but I show you the notification that pass through the end process. The question is every print job I would choose the proper gcode for each mode (marlin & touch mode).  

[gcode opt1.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3895652/gcode.opt1.txt)

bigtreetech, 2019-11-29:
Hi. I have confirmed this problem. This is a bug. The screen is not compatible with parameters such as ‘B’ ‘F’ ‘R’ of “M104” “M140” “M109” “M190”. I will fix it as soon as possible

bigtreetech, 2019-11-29:
very thanks for you feedback

Interplain, 2020-01-06:
This issue is same as i am getting with current TFT 35 v2.0.25 with SKR 1.3

deebeo-blip, 2020-02-04:
Me as well. Can you tell, when a new Version will be released. I am currently on 2.0.25.1 with SKR v1.3 and I am experiencing "command unknown" Error messages on the display resulting in Y Layer Shifts while printing. Thank you,

bigtreetech, 2020-03-20:
It should be fixed in #402, so I close here

closed 2020-03-20 by bigtreetech