bigtreetech/BIGTREETECH-TouchScreenFirmware #196: Gcode list with icon display proposal

GenaUser, 2019-11-27:
I propose this to try to have both advantages (long file name and model preview icon).
First step : preselect gcode line with a preview icon displaying in the right bottom corner.
Second step : click on the icon for start the printing.

[image]

GenaUser, 2019-11-27:
Other proposals:
Click on gcode line, and displaying a popup window showing a large icon label with printing confirmation.

dexter323i, 2019-11-28:
I agree with both! A confirm option or popup would be highly appreciated! The lines are small, it is very easy to miss the click, and it starts to print inmediately.

HappyCodingRobot, 2019-12-01:
Some idea how we get the preview icon?
Post processing after slicer? Plug-in for slicer that adds this to the gcode meta information? Would this work for PrusaSlicer?

Hukuma1, 2019-12-05:
Looks like we all are on the same page on what could make things better. :)

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/118#issuecomment-562280265

GenaUser, 2019-12-06:
@Hukuma1, I totally agree with you and your visual proposition.

closed 2020-08-14 by bigtreetech