bigtreetech/BIGTREETECH-TouchScreenFirmware #197: TFT35 V2 after bootloader update screen shows "illegal user App1!"

acidcoolxxx, 2019-11-27:
Hi,

I did the bootloader update for uploading new Firmware. But after i flashed the BIQU_TFT35_V2.0_bootloader.hex i get the error: Illegal user App1!

This is my log for the flashing:
...Delay 100ms
DTR be Low(-3--12V),Reset Released
RTS keep High
Connectting...5, Received:1F 1F 
Connect Ok @COM6@115200bps,@844ms
BootLoader Version:2.2
PID:00000414  STM32F10xxx_High-density
OptionBytes readout:
A55AFF00FF00FF00FF00FF00FF00FF00
96bit unique device ID:
[39FFD7054E59383418800543]
[05D7FF39 3438594E 43058018]
FLASH ROM size:256KB
SRAM size:65535KB(For reference only!!!)
@906ms,Ready for Program
Write 10KB Ok,100%,@5297ms
Write Option Bytes: A5 5A FF 00 FF 00 FF 00 FF 00 FF 00 FF 00 FF 00
Write Option Bytes Ok
OptionBytes JustWrited:
A55AFF00FF00FF00FF00FF00FF00FF00
Go from 08000000 Failed...Maybe Writed OptionBytes just before!!!
www.mcuisp.com(Versatile Handy Programmer EP968,World's first):Mission Complete,Anything Ok!!!

Try to upload new firmware with font. But the tft dont even try to update somehing it only shows the error code

acidcoolxxx, 2019-11-27:
I managed to update with a new sd card. But now my touch dont work. Do i need to change something in the firmware?

dineshvyas, 2019-11-27:
it  is normal.
now you can update firmware by sdcard.

closed 2019-11-28 by acidcoolxxx

reipet, 2020-04-17:
Can you tell us how you fixed this Issue? Im having the same one.
Initially it worked but when I updated the SKR this message shows up

Cheo7811, 2020-04-25:
I have got exactly the same problem, TFT32 v2.0 will not update firmware from card, updated firmware and it ran exactly as per the instructions as shown in the post by the OP and when the programmer ended the display will show "illegal user App1!"
App1 in this display as I understand is the Display firmware but it does not update firmware from the card.
I tried re-flashing the bootloader but it will not longer respond to the the programmer.
Any help will be greatly appreciated.

reipet, 2020-04-26:
This worked for me:
Prepare the SD-Card as stated in the update instructions but also copy the buildroot folder to the SD -Card as well. Then the Screen will update again and it should work. Not sure what causes this issue.
I tried to update the icons only then Im having the same message again. Seems very unstable, buggy Firmware.

Oburi85, 2020-04-28:
Got the same problem. The "BIQU_TFT35_App1_V2.0.26.bin" worked for me.

Rockerbilly, 2020-05-18:
Now i have fixed the Problem
You have to copy the complete folder TFT35 and your Firmware.bin to the SD card, then press the reset button and the update starts as desired and the display works as it should
Hope I could help

telephat, 2022-03-21:
> I did the bootloader update for uploading new Firmware. But after i flashed the BIQU_TFT35_V2.0_bootloader.hex i get the error: Illegal user App1!

You have to copy TFT folder from inside of THEME_<theme of your choice> to SD card
**And you have to copy .bin firmware again**, because it becomes useless and corrupt after first, unsucessfull upload.

Poco9964, 2023-05-05:
Avoid BTT the products suck, the code is worse and the support is pointless.