bigtreetech/BIGTREETECH-TouchScreenFirmware #198: Change filament

guzio14, 2019-11-27:
Hello. Will the "change filament" option (M600) be available? Tft35 v2.0

HappyCodingRobot, 2019-11-27:
Atm you can use it via the custom gcode option. (in case you you meant to trigger it manually)

guzio14, 2019-11-27:
It was more about Advanced Pause features, with the option change filament.

thisiskeithb, 2019-11-27:
It'd make the most sense to add this as a configuration option since `ADVANCED_PAUSE_FEATURE` is not enabled in Marlin by default from many printer manufacturers.

guzio14, 2019-11-28:
> It'd make the most sense to add this as a configuration option since `ADVANCED_PAUSE_FEATURE` is not enabled in Marlin by default from many printer manufacturers.

That's what it's all about 😃
Właśnie o to chodzi

Taomyn, 2020-03-30:
Yes please, I hate that it's easier for me to revert back to Marlin mode to get smart filament load/unload than use the touch screen. I have a TF35_V3.

closed 2020-08-17 by bigtreetech