bigtreetech/BIGTREETECH-TouchScreenFirmware #200: unified interface : long filename not display when print from internal SD

melufre, 2019-11-27:
1st thank you continue to improve the TFT firmware, this version (25) is very nice.

the main board is an SKR1.3 with marlin 2.0.x 2019-11-17
using the same SD card 
 - on SD TFT : long file name are correctly displayed
 - on internal SD (SKR1.3) : file name are short (with ~1)

guruathwal, 2019-11-28:
The SKR itself is sending the short file names.

Powerprobot, 2019-11-28:
You must compile marlin with this line 
```
#define LONG_FILENAME_HOST_SUPPORT
```
in CONFIGURATION_ADV.h

macasero, 2019-11-30:
> You must compile marlin with this line
> 
> ```
> #define LONG_FILENAME_HOST_SUPPORT
> ```
> 
> in CONFIGURATION_ADV.h

Already have that uncommented and it doesn´t work. Still short names

Powerprobot, 2019-12-01:
ok. I see. Marlin handle long filenames other than the TFT. The TFT need additional coding to receive, store and view long filenames.
Read the Wiki for Marlin "M33"
@bigtreetech please add the long filenames. My c++ knowing is too small to code this.

macasero, 2019-12-03:
No, TFT doesn´t need aditional coding, because raspberry octoprint also sees ONBOAD SD with short names, even with the TFT off, so It´s not a TFT problem, instead it´s a SKR problem

melufre, 2019-12-03:
i try to change the M20 to report long file name on marlin , display is good on TFT , but print is not started (M23) . it is because marlin work with short (DOS) filename . changing filename handling in Marlin seem to be too complex (at least for me)
So if think the proposition  of @Powerprobot  to use M33(get long filename) in TFT sw is a good proposition ,  @bigtreetech please call M33(get long filename) when you list internal SD files

melufre, 2019-12-06:
i have create pull request https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/224
with this you should be able to display long file name when using onboard SD
note that folder are still display with short name

macasero, 2019-12-16:
I tested your branch and It works, but this icons don´t work fine
[https://imgur.com/hRLKJ3A](url)

melufre, 2019-12-16:
Yes i quite strange it affect only those icon , probably a memory overlap

macasero, 2019-12-17:
it seams that the is some character that the font isn´t allow to print

closed 2020-08-14 by bigtreetech

blueeagle69, 2020-08-24:
I have enabled long filename support in Marlin etc.
I correctly see long filenames when browsing gcode (on board SD). However when a print is in progress the filename that displays on the main screen during the print reverts back to 8.3 format.

Any ideas?