bigtreetech/BIGTREETECH-TouchScreenFirmware #204: TFT35 V2 Touch not working

acidcoolxxx, 2019-11-29:
Hi

after firmware flash my Touch have no respond. I insert the option to Force-Reset touch screen calibration. I can calibrate it but after booting it there is still no respond. Do i need to change something in the Settings, because i have no rotary encoder.

guruathwal, 2019-12-02:
Try reset again and do not remove the sd card until calibration is done.
Aftrr it remove sd card and press reset button

phcay, 2019-12-04:
I had the same problem yesterday and I saw that it was missing the writing of the setting after the calibration.
I reported the that to guruathwal. Future commit will have to work.
If you want to solve your problem now, make this small change.

in boot.c add bold line on scanResetDir() function and compile \:

void scanResetDir(void)
{
  FIL resetfile;
  if (f_open(&resetfile, TFT_RESET_FILE, FA_OPEN_EXISTING | FA_READ) == FR_OK)
  {
    f_close(&resetfile);
    f_rename(TFT_RESET_FILE, TFT_RESET_FILE ".DONE");
    infoSettingsReset();
    TSC_Calibration();
    **storePara();**
  }
}

guruathwal, 2019-12-05:
This has been fixed through #215 .
Waiting for @bigtreetech to merge

closed 2019-12-05 by acidcoolxxx