bigtreetech/BIGTREETECH-TouchScreenFirmware #208: [BUG] [ABL] Bad display for negative values

GregSKR, 2019-12-01:
In Unified Menu, when going in ABL menu then [Offset] or [Babystep], negatives values are not displayed correctly.
eg: -1.26 is displayed  .260 (no negative sign, no value before the point)

Hukuma1, 2019-12-02:
[image]

Got the same thing. -1.75 shows up as .750

TFT35 V3 with latest firmware. Was not brave enough to test a print to see if it's just cosmetic bug. :P

guruathwal, 2019-12-02:
Its just a glitch.. it will not affect your print..

GregSKR, 2019-12-02:
Yes, just a glitch for sure. But very confusing when using those functions!
I hope a programmer will give a look at it...

billydragon, 2019-12-03:
I fixed this by edit  ProbeOffset.c :

#define PROBE_OFFSET_MAX_VALUE 20.0f             //My Chiron Probe Offset large then 16mm
#define PROBE_OFFSET_MIN_VALUE -20.0f           //

void showProbeOffset(void)
{
  GUI_DispFloat(CENTER_X - 5 * BYTE_WIDTH / 2, CENTER_Y, probe_offset_value, 3, 2, RIGHT);
}
void probeOffsetReDraw(void)
{
  GUI_DispFloat(CENTER_X - 5 * BYTE_WIDTH / 2, CENTER_Y, probe_offset_value, 3, 2, RIGHT);
}

and Babystep.c:

#define BABYSTEP_MAX_VALUE 5.0f
#define BABYSTEP_MIN_VALUE -5.0f

void showBabyStep(void)
{
  GUI_DispFloat(CENTER_X - 5*BYTE_WIDTH/2, CENTER_Y, baby_step_value, 3, 2, RIGHT);
}
void babyStepReDraw(void)
{
  GUI_DispFloat(CENTER_X - 5*BYTE_WIDTH/2, CENTER_Y, baby_step_value, 3, 2, RIGHT);
}

re- compile firmware then this gone, display correct negative value.

extesy, 2019-12-03:
@billydragon can you send a pull request for this fix?

billydragon, 2019-12-03:
I'm new in github, i don't know how to pull request do

extesy, 2019-12-03:
@billydragon https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request

billydragon, 2019-12-03:
Pull request done

GregSKR, 2019-12-03:
Great, thanks guys!

closed 2019-12-03 by GregSKR