bigtreetech/BIGTREETECH-TouchScreenFirmware #211: [BUG] TFT35 v2 and SKR v1.3 M0/M1 commands cause endless Confirm prompts

kepbrown, 2019-12-03:
Screen: TFT35 V2 
Firmware: TFT35_V2.0.24.2

Board: SKR v1.3
Firmware: Marlin bugfix-2.0.x

### Description
Executing GCode from the TFT on board SD card slot that contains an M0 command causes an endless loop of screens displaying 
echo: 
busy: paused for user 

If I click the Confirm button the screen clears then pops back up. The only way to stop is to reset the screen.

I haven't attempted to run it from the board SD slot. I assume it would work the same way.

### Steps to reproduce
Print a gcode file with the following commands

;FLAVOR:Marlin
G28 ;Home
G1 Z10 ; Lift Z axis
G1 X32 Y35 ; Move to Position 1
G1 Z0
M0 Test Output ; Pause print
M106 S0 ; turn off cooling fan
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M82 ;absolute extrusion mode
M104 S0
;End of Gcode

**Expected behavior**
The prompt comes up but once Continue is pressed it moves on to the next step

**Actual behavior**
Endless loop

### Additional Information
The file being ran is attached. txt extension added to allow it to be uploaded.
[pause_test.gcode.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3914766/pause_test.gcode.txt)

guruathwal, 2019-12-04:
with M0 command the printer will wait for the user to press continue button on the **LCD** not the **TFT**. untill you press it the printer will send the busy command to the TFT, which will result in the same thing you are observing.

Doenermaker, 2019-12-05:
Wouldn't it be possible to send M108 when clicking 'confirm'? M108 needs to be enabled in Marlin, but to my understanding it should work as well as pressing the lcd button.
I have the same problem after M125 in the gcode. My guess is M600 would also behave the same, haven't tried tbh. My TFT35v1.2 is the only display connected, so pressing the lcd button isn't an option for me.

kepbrown, 2019-12-06:
I have the same issue with only having the TFT connected, I can't get my LCD to work with my SKR v1.3. It would be nice to be able to find a solution that doesn't require the LCD.

moag1000, 2019-12-17:
Could you reconsider that this IS actually an issue? I don't have an LCD just like kepbrown and I've tried to send a M108 and M867 S0 (with custom button because the "busy: Pause for user" keeps interrupting and deleting G-code) which does not go through.
I could live with that at least, if it's possible.

Interplain, 2020-01-04:
Same problem with same Firmware, i got other issues. Think this board is full of bugs. 
Screen: TFT35 V2
Firmware: TFT35_V2.0.24.2
no USB Detection either.

guruathwal, 2020-01-05:
where is the gcode file you are trying to print located? on the Main Board SDcard or TFT sd card?

Interplain, 2020-01-05:
The Gcode that I am printing from is located on the SD card in the TFT screen. I have not tried it from the TF card because I have no connection via USB to the SKR v1.3 Motherboard. I could take the TF card out an put a file on that but have not yet, but i suspect that it will be fine. I did however have connection to the motherboard via USB and it printed fine. The motherboard failed while in the process of printing via USB with Simplify3D and after rebooting it failed to show up on Win10 computer as a Device. I had to use the SD card because no other way of getting Gcode into the MBoard. I seems to constantly flash up with echo, with a confirm button. It seems to print ok, but to change settings they revert back to what they are set in the Gcode. So I have issues with main board too, but that's another issue. I have got another board coming to sort that, I think its faulty USB chip. I have formatted the TF card an put the firmware back on and it still didnt show. So I all I have two issues.. But do have great expectations for the thing, it just I am frustrated with the teething problems I currently have. I you need any more information I am will to chip in with the topic that this currently addressing. The Screen.

Interplain, 2020-01-06:
Tried the TF card on the board an it will not read the card says SD error no found, Think i have a faulty Boot loader but not sure how to flash new one to it, or where to get it from. Well annoying..  Got the printer working like a dream. so when new board comes i know main board will be fine.

Interplain, 2020-01-13:
When is this going to be fixed?

guruathwal, 2020-01-14:
@Interplain already fixed in #297

Interplain, 2020-01-14:
The Gcode is located on the SD card in the TFT Screen.  When I print from the SD Card I get a continuous prompt asking me to confirm this is only in the Unified version it doesn't do it on the Orginal. It's a prompt with echo: and something like G28 and confirm. And when it prints shows all coordinates and confirm with each move.

guruathwal, 2020-01-14:
@Interplain  I am not able to understand your problem ... Did you tried #297 ??

Interplain, 2020-01-14:
I will try that later, thanks. Will enabling EMERGENCY_PARSER stop the Echo box showing. Sorry didn't know that it had been investigated.

guruathwal, 2020-01-14:
yes you will have to enable EMERGENCY_PARSER.
currently popup will not be displayed if it receives "paused for user" text from printer

Interplain, 2020-01-14:
Enabled the above and it still shows, blocking off access to underlying buttons.

[image]

Are we talking about the same issue here?

ipatch, 2020-01-14:
@Interplain 

you are compiling this firmware from source, right?

and you did merge the PR @guruathwal submitted upstream to your local fork, right?

As of right now, you have to manually merge the PR that @guruathwal sent upstream as it hasn't been merged into the master branch yet.

If this all stuff you've already done, please forgive me, as I didn't see any mention of this before in the thread.

Interplain, 2020-01-16:
No, I have not compiled this. I thought it had been fixed in the firmware for the TFT 35 2.0.25-1
I have still got the issue, so I guess I will have to wait until its been put in a new firmware.

Doenermaker, 2020-01-16:
The PR was merged into the master branch 4 hours ago, so you can download the latest files, compile the firmware yourself, and see if your problem is fixed @Interplain

Interplain, 2020-01-16:
That would be great if I knew what to do, it would be nice to do it but not sure where to start. I am familiar with Marlin but not had any experience with the screen I have only had it a week or two.

CRCinAU, 2020-01-21:
Ok - so I've just looked at this a little closer after sending an M701 to load filament and not being able to hit the OK button.

It seems that you SHOULD be able to send an M108 to continue, M112 to cancel (STOP).

Problem is, this needs EMERGENCY_PARSER feature in Marlin and:
```
Marlin/src/HAL/HAL_STM32F1/../../inc/../HAL/HAL_STM32F1/inc/SanityCheck.h:29:4: error: #error "EMERGENCY_PARSER is not yet implemented for STM32F1. Disable EMERGENCY_PARSER to continue."                                                    
   #error "EMERGENCY_PARSER is not yet implemented for STM32F1. Disable EMERGENCY_PARSER to continue."                                                                                                                                        
    ^~~~~                                                                                                                                                                                                                                     
```

As such, these options won't work anyway :(

closed 2020-08-14 by bigtreetech