bigtreetech/BIGTREETECH-TouchScreenFirmware #218: [BUG] Cooldown after print, without gcode command

dexter323i, 2019-12-05:
### Description

MainBoard is SKR Mini E3 V1.2 With latest Marlin 2.0 from BTT GitHub repo
TFT35 V3.0 FW is V3.0.25

I installed a BMG Clone extruder in place of original Ender 3 extruder, so it is still a bowden configuration. I can't load / unload the filament by hand, because the extruder gears so tight, therefore I need a filament load / unload gcode script. I use a slightly modified version what Bondtech offered. The problem is, when I print the load script from the printer, the TFT35 in Marlin mode, then everything works as it should. After the print, the nozzle keeps the 210 celsius temp.
But when I print it from TFT35 onboard SD port, after the print, the TFT turns off all the heating, and everything is starts to cool down. It is a problem, because after the filament loading, I want to start the print, but I have to wait until the bed, and the hotend heats up again!

Please, don't turn it off. Or at least, make it configurable!

### Steps to reproduce

1. Put the attached file to an sd card, renamed to *.gcode
2. Turn on the printer in Touch mode, with the sd card in the display's SD card port
3. Not necessary to put filament in
4. Start to print the file
5. Wait until nozzle heats up to 210c
6. Check the extruder starts to turn quickly
7. Realize the screen kind of freezes. Nothing changes, not even the elapsed time
8. Then the extruder starts to extrude slower, the screen still frozen
9. When the extruder stops, the screen recovers, the nozzle (and bed also) starts to cool down. (Nothing like this in the gcode.)

### Steps to reproduce 2
You can find a huge amount of lorem ipsum at the end of gcode. It is because I've read, the TFT35 has problems with gcodes under 2mb. In my case, it is worse:

1. Put the attached file to an sd card, renamed to *.gcode
2. Delete the huge lorem ipsum comment from the end of the file
3. Turn on the printer in Touch mode, with the sd card in the display's SD card port
4. Put PLA filament in. It's end should be very close tho the extruder gear
5. Start to print the file
6. Wait until nozzle heats up to 210c
7. Check the extruder starts to turn quickly
8. Realize the screen kind of freezes. Nothing changes, not even the elapsed time
9. Then the extruder starts to extrude slower, the screen still frozen
10. After few seconds, the extruded filament (what comes out from the nozzle)
 starts to come out slower, then it curves back up. Showing signs of a massive cold extrusion. 
11. When the extruder stops, the screen recovers, the nozzle shows around 160c and dropping! It means, the hotend heat was turned off during the print! It is a HUGE problem.

**Expected behavior**
Nozzle and bed should keep the themp after print, because there was no cooldown command in gcode

**Actual behavior**
Nozzle and bed starts to cool down at the end of the print, OR during the print. (Reproduce 2)

### Additional Information
gcode file attached:
[LoadFilament.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/3929535/LoadFilament.txt)

guruathwal, 2019-12-06:
hi, Instead of using the .gcode file to perform these actions you need to use the Custom gcode option available in the `Configuration.h` file.

dexter323i, 2019-12-06:
Okay. The custom gcode kind of solved the problem. Now, sometimes it shows an **Unknown Command "MM105"** message, when I start my custom scripts. (No! Nothing like this in the gcodes)  But now it works.

Are you willing to solve the original issue about turning down the heaing? Or no, and I should close this issue?

`#define CUSTOM_1_LABEL "LOAD"`
`#define CUSTOM_1_GCODE "G21 \nG91 \nM104 S210 \nM109 S210 \nG1 E350 F1800 \nG1 E100 F200 \nM400 \nM300 S440 P200 \nM300 S660 P250\n"`
`#define CUSTOM_2_LABEL "UNLOAD"`
`#define CUSTOM_2_GCODE "G21 \nG91 \nM104 S210 \nM109 S210 \nG1 E10 F100 \nG1 E-500 F1800 \nM104 S0 \nM400 \nM300 S440 P200 \nM300 S660 P250\n"`

guruathwal, 2019-12-06:
The heater turning off after print through TFT is a safety feature to ensure heaters are turned of if it is missed in gcode file. Will try to add option in setting to turn it on /off.
"MM105" is strange.
You should avoid putting space before the line break `\n` wherever possible.
You can close the issue for now.

dexter323i, 2019-12-06:
Okay, thank you for your help!

closed 2019-12-06 by dexter323i