bigtreetech/BIGTREETECH-TouchScreenFirmware #230: BIQU TF35 V1.2 Firmware Update don't work

nolsnols, 2019-12-08:
I have a BIQU TFT35 V1.2
Board: BIQU_TFT35 V1.2
Firmware: BIQU_TFT35_V1.2.6

I cant update the firmware via SD Card. I downloaded the firmware copy the files for the V1.2 to SD Card, but it only updates the bmp and fonts.

I renamed the .bin file to: firmware.bin, BIQU_TFT35.bin BIGTREETECH_TFT35.bin
But it never updates the firmware. Only the bmp and fonts.

Tried with 5 different SD Cards. Printing works well, Updateing the Fonts and BMPs works.

How can I update the firmware for the TFT35 V1.2?

moag1000, 2019-12-17:
Did you try the original name "BIQU_TFT35_V1.2.25.bin" yet?

nolsnols, 2019-12-17:
I have a 2GB SD Card, 8GB SHDC, 16GB SDHC, 64GB SDHC.
Formatted with SD Formatter on Mac (fat32).

Tried Firmware: 
BIQU_TFT35_V1.2.11.bin
BIQU_TFT35_V1.2.12.bin
BIQU_TFT35_V1.2.25.bin

with original filenames.
Than renamed the .bins to firmware.bin, BIQU_TFT35.bin, BIGTREETECH_TFT35.bin, BIQU_TFT.bin

Nothing will flash the TFT...
I have to flash the bootloader? or can I update via PC?

guruathwal, 2020-01-03:
in which folder arre you copying the firmware on the sd card?

nolsnols, 2020-01-06:
No folder. Format is Fat32 and i copy it to the root.

askuechler, 2020-01-12:
I second that problem. Also Firmware 1.2.6 on the same hardware and I tried all available firmware variants. The only thing working is the update of the icons and fonts but the firmware itself is not changing.

nolsnols, 2020-02-15:
So, no solution to update the V1.2.6 TFT Firmware?
Maybe a serial loader or something else?

guruathwal, 2020-02-16:
You can use `stm32 flash loder demostrator` and flash the firmware using rs232 port on the tft. You need to google it to know the full process.

Pixel-Soup, 2020-03-05:
My suggestion is to use a SD card, (not SDHC)  formatted to FAT  (not FAT32).  It works on my tft24.  I think the bootloader cannot read the newer standard cards.

JustGand, 2020-03-13:
I was experiencing the same problem (unable to update from V1.2.6), and renaming the BIQU_TFT35_V1.2.25.2.bin firmware to BIQU_TFT35.bin worked !

askuechler, 2020-03-21:
Hi,
using an old card, FAT and renaming the file as JustGand suggested did the
trick for me too. Thanks very much.

gordo3di, 2020-09-05:
@nolsnols Can you please close this ticket. Thanks.

oldman4U, 2020-09-07:
@gordo3di 

Your help is very welcome. I appreciate!;-)

oldman4U, 2020-09-11:
@bigtreetech 

Please close this ticket because the topic is solved and the user does not answer anymore

Thank you

gordo3di, 2020-09-12:
@oldman4U Keep up the good fight... I don't get why people or bigtreetech wont just close some of these

oldman4U, 2020-09-12:
Hi

Asked to close 20+ tickets. 3 closed and 6 new tickets means a total of
plus 3 tickets. Sometimes it is really frustrating.

🥺

closed 2020-10-10 by nolsnols