bigtreetech/BIGTREETECH-TouchScreenFirmware #231: UART3 and UART4 not correctly work

billydragon, 2019-12-08:
I have BTT-SKR V1.3 and TFT35 V3.0 connected. I want to using UART3 or UART4 for directly TX and RX 3 wires to Octoprint Pi. This UART some time work and some time not work. I don't know how to config for UART3~4 work properly. Can anybody help?

Aleks-MO, 2019-12-12:
This UART some time work and some time not work?

ipatch, 2019-12-12:
take a look at how i solved this issue 

[here](https://github.com/ipatch/Marlin/tree/ipatch/build-marlin-2.x/media#solved--octoprint-reconnection-issues)

basically if your not using the wifi module you can use the rx/tx pins from it. And there is one setting in octoprint that needs to be changed, I included pictures of both in the link.

billydragon, 2019-12-12:
I found that:
UARTs and Wifi only work when in LCD Touch Mode, in Simulator 12864 not work.
I have Wifi module connected and it work. UART3 and UART4 work only when you have connect to wifi first then Octopi using UART3 or UART4 will work after that. 
I try remove Wifi module and using RX/TX from this wifi header but octoprint don't work.

ipatch, 2019-12-12:
so there's a lot of moving parts when connecting the raspberry pi to one of these boards or even screens that support UART.

Personally, I don't have a BTT TFT35v3 but rather I have the TFT35v2, and it did take me a while to get everything setup.

I'm a little confused as to why you're using a wifi module on your screen, when you can use octoprint on the pi, and then just connect your pi to your LAN ie. your home wifi network, or where ever you have your Pi and printer located.

I'm assuming your using the GPIO pins on the Pi, no?

What version is your Pi? ie. 3, 3b, 3b+, 4, 4b?

And have you tested connecting octoprint to the board itself, ie. taking the screen out of the equation?

1. Can you connect octoprint to your board via the USB port on your board?
> assuming you a SKR board, no?
2. Are you compiling marlin from source, or are you running a prebuilt / precompiled version of marlin?
3. What OS are you running on your Pi?
	a. is the OS Raspbian? with octoprint installed on top?
	b. or is it the Octopi image itself?

The more details you can provide about your setup, the more likely people will be willing to provide information, which hopefully solve your problem.

The less information you provide the less likely someone will want to take the time to aid you with your issue.

billydragon, 2019-12-13:
I try wifi module for test only. i can remove wifi module if i can connect Octopi directly rx/tx pins and free USB port for PC using upload firmware and files to SD card.
I have Pi3B+. My 3D printer is Anycubic Chiron. I make config, compiler from source code Marlin bugfix 2.0.x, also compiled FW for TXT35 V3 from source code . every things working fine except UART connect.
I will try again using rx/tx of wifi module to connect to Octoprint again.

EsQueue81, 2019-12-29:
Sorry to necro this but what's the update on this? I have a raspberry pi 3b running octopi, a skr 1.3 and tft30 v3.0 with to WiFi module installed. I initially tried to connect the pi using the 3 wires connected to the AUx-1 of the board but it never worked and checking out the board's schematic shows that The AUX-1 RX and TX connects to the TFT pins with a resistor.  I then connected it to the uart 3 or 4 on the TFT35 v3.0 screen it it worked instantly.  

Will this preform at least as good as the USB connection?

gilliands, 2020-01-06:
I've done some experiments on a TFT24 and an older TFT35 V1.2 and added a second serial to both to support an ESP module. No reason why this could not be a PI instead.  All I did was define the serial ports in the variant.h file and defined the next serial port and assigned the USART that mapped to the pins available eg:

#elif defined(TFT24_V1_1)
  #define SERIAL_PORT   _USART1  //default usart port
  #define SERIAL_PORT_2 _USART3  // Added this port to map to PB10,11 for ESP module

In the case of the TFT24, the signals pop up on EXP1 PB10 and PB11 and on the TFT35 v1.2 it's possible to map USART2 and fish the PA2, PA3 signals from the corner pins 25 and 26.

EsQueue81, 2020-01-07:
> I've done some experiments on a TFT24 and an older TFT35 V1.2 and added a second serial to both to support an ESP module. No reason why this could not be a PI instead. All I did was define the [...]

I made that post prematurely. Connector labeled UART 3 allowed my Pi to connect but it never gets the M105 commands and times out after 30 seconds if I don't keep sending commands manually. The UART 4 port sends the temperatures just fine and looks like it will work great but never moves past the connecting state. It doesn't respond to any manually input codes via octoprint.  

Thanks for this post, it will give me something to play around with. I'll also copy your quote to another thread where someone else is having my exact same issue.

hapklaar, 2020-01-26:
Keep us posted about your progress please. I was never able to successfully use OctoPrint (from Pi) to the TFT35 in combination with an SKR 1.3, and also get useful statistics on the TFT status screen.

Still looking for a way to do this.

EsQueue81, 2020-01-27:
> Keep us posted about your progress please. I was never able to successfully use OctoPrint (from Pi) to the TFT35 in combination with an SKR 1.3, and also get useful statistics on the TFT status screen.
> 
> Still looking for a way to do this.

Unfortunately, Bigtreetech releases devices with the absolute minimum amount of information with them. I've come to the conclusion that the only way to solve problems is to be a C/C++ programming and figure it out by yourself.

viciouslancer, 2020-01-30:
@ipatch can you please provide the information on how to use WiFi RX/TX pins? I have a SKR 1.4 Turbo but I imagine the info is relatively the same. Sadly, the link to your fix is not working.

ipatch, 2020-01-31:
@viciouslancer 

updated my previous link, and all my notes building marlin with big tree tech hardware can be found [here](https://github.com/ipatch/Marlin/tree/834266a73193f5c803502930b0d7de142e9325bf/media)

brunnels, 2020-07-17:
> I've done some experiments on a TFT24 and an older TFT35 V1.2 and added a second serial to both to support an ESP module. No reason why this could not be a PI instead. All I did was define the [...]

@gilliands I did the same with TFT24 and also tried to put the filament runout on the unused PB14 but I'm not getting anything on PB10 and PB11.  I tried swapping pins around and multiple USB TTL adapters.  I can see the serial output of my SKR board if I touch the RX of the TTL adapter to the TX going to the TFT24 so I know it's not the TTL adapter.

I also tried both 
#define SERIAL_PORT_2 _USART3
and 
#define SERIAL_PORT_3 _USART3

In the config.ini I also tried setting `serial_always_on:1` but still nothing.

Any ideas?

closed 2020-08-14 by bigtreetech