bigtreetech/BIGTREETECH-TouchScreenFirmware #233: [BUG] Marlin K-Factor calibration broken

Hukuma1, 2019-12-10:
Using standard Marlin code generator to tune k-factor settings. The bed temp shows it trying to be set to 115, instead of 60.

### Steps to reproduce

1. http://marlinfw.org/tools/lin_advance/k-factor.html
2. Generate code for your machine settings.
3. Print it.

**Expected behavior**
Bed temp raises to 60 and starts print.

**Actual behavior**
Bed tries to raise temp to 115.

### Additional Information
https://pastebin.com/12LhRcfA

guruathwal, 2019-12-20:
That's strange. Did you set preheat before it?

Hukuma1, 2019-12-20:
Nope. And definitely never to a weird number like 115.

If you run it, does it work fine? :/

guruathwal, 2019-12-20:
Not tried it yet.. I just saw this issue now..😊

Hukuma1, 2019-12-20:
Just tried running same code in Marlin mode and it worked as expected. So it's definitely an issue with Touch Mode/firmware.

closed 2020-08-14 by bigtreetech