bigtreetech/BIGTREETECH-TouchScreenFirmware #236: strength looking icons TFT35 v3.0

andrzej1974, 2019-12-12:
Hello.  
[image]
After compilation in VSC.   - Tft35 v3.0, bigtreetech software and uploading to Display i have in the Print menu  in the file selection list strange squares  on the icons with arrows.   if i upload firmware to the display  i putt ofcourse TFT35 catalog with BMP and fonts  on sd card.  I use  unified firmware. Somebody tell how delete this squares?

carl1961, 2019-12-12:
have you tried the already compiled bin files? to test

andrzej1974, 2019-12-12:
i try and is ok but I have to make modification in firmware, because dont
see long filename from onboard sd card. But if I compile original without
modification also is the same issue.

carl1961, 2019-12-13:
to me it looks like pictures on top of the other

guruathwal, 2019-12-13:
> i try and is ok but I have to make modification in firmware, because dont see long filename from onboard sd card. But if I compile original without modification also is the same issue. 

@andrzej1974  what code did you modified? 
**There are some variable that do not need to be changed in any case!!** 
Can you share the changes you made?

andrzej1974, 2019-12-13:
This one
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/235
But before modification i allready have this issue, from first compilation
without touching  anything. Just fresh compilation and upload directly to
TFT this square was visible

guruathwal, 2019-12-13:
> This one #235 But before modification i allready have this issue, from first compilation without touching anything. Just fresh compilation and upload directly to TFT this square was visible

ok... looks like it got messed up after #205

andrzej1974, 2019-12-13:
I have already implemented it drom (#236)

andrzej1974, 2019-12-13:
sorry this one (#205)

pt., 13 gru 2019, 20:22 użytkownik Andrzej <andacz6@gmail.com> napisał:

guruathwal, 2019-12-14:
> I have already implemented (#205) 

@andrzej1974  #205 messed up the list view. you need to use the source before this commit for now.

guruathwal, 2019-12-19:
Fixed via PR #246

guruathwal, 2020-01-03:
@andrzej1974  please close the issue

closed 2020-01-03 by andrzej1974