bigtreetech/BIGTREETECH-TouchScreenFirmware #239: [QUESTION] Who has TFT24 or TFT28?

guruathwal, 2019-12-15:
Who has TFT24/28 and can test the firmware for checking the UI? need to check the new toggle interface before I create the PR?
here: https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware-MOD

post photo or video if possible.

oldman4U, 2019-12-16:
Hello guruathwal.

I have an SKR Pro v1.1 with a TFT28 running the latest firmware and this combination is not in production but only for testing. How can i help you? Is there a particular area i should check or just upload and see if everything seems to work?

Would be a pleasure for me to give back a little bit;-)

Georg

guruathwal, 2019-12-16:
@oldman4U  thanks.. all I need to test is the UI for TFT in the `setting menu->feature settings` & `setting menu->printer settings>custom menu`.
you can download the firmware for your variant from here: https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware-MOD

oldman4U, 2019-12-16:
I will do so and let you know in approximately 10h from now.

Kind regards

oldman4U, 2019-12-16:
Hi again.

So I downloaded the files from the link above, copied the BIQU_TFT28_V1.0.25.1.bin and the TFT 28 folder onto an SD card, inserted the card into the reader of the display and turned everything on. It uploaded the firmware, then the font and the bitmaps. I got the Ready in the upper left corner so this looks ok for me. Attached are some screenshots for verification that what I see is really the expected result. I can press the items in the list and get gCodes back, even it says unknown command when I try to release the SD card. Pressing "Hide Terminal ACK" modes the red button and makes it green and vice versa. "Auto Power" stays red (guess something in Marlin is missing;-))

Hope this helps. Please let me know if you need anything else

Thank you
[image]
[image]
[image]

guruathwal, 2019-12-17:
@oldman4U  thanks for the review..
>Pressing "Hide Terminal ACK" modes the red button and makes it green and vice versa. "Auto Power" stays red (guess something in Marlin is missing;-))

its not in marlin, some variant of the TFT does not support runout sensor Pins and auto shutdown pins, it need to be fixed in TFT firmware..

oldman4U, 2019-12-18:
OK, understand.

Please let me know if there is anything else i can help with.

;-)

oldman4U, 2019-12-19:
Hello guruathwal.

May I ask you 3 questions please?

1. Is it always like this, that you ask for help but not a single user is there to help you?
2. Would it be useful for you, if I would buy a TFT24, so I can test it in the future?
3. Maybe you have seen the other ticket #243. Has this part of the firmware something to do with it and if yes, how can I activate this and what's needed for it?

 * Enable gocde files list mode
 * It is friendly to display long file name, but the model preview feature is not available
 * Disable this if you want to use the model preview feature

Thank you

guruathwal, 2019-12-20:
@oldman4U  Thanks
1. Most of the time. I try to help wherever I know anything about the issue. 
2. That would be nice. But the tft24 and 28 have the same screen resolution so testing on one of them would be enough.
3. Displaying the preview while printing is not yet implemented. You can disable list view if you want to display model preview in file viewer, but you will have to manually create preview bmp file for each gcode file with application provided by BTT. As far as I have seen yet in almost all the display available, the model preview is to be created manually. I don't think its worth to do all those procedure just to preview the image on the icon which will be visible Only for 5 seconds, and deleted after printing.

Hukuma1, 2019-12-20:
BTT needs to write a Cura plugin that will export a snapshot of model on save of file. Creating the previews right now is too cumbersome.

oldman4U, 2019-12-20:
Good morning.

Ok, so i will spend my money for something else;-)

Thank you for the answer, also to you Hukuma1.

Happy holidays

closed 2019-12-20 by guruathwal

ETE-Design, 2020-01-26:
@guruathwal Would it be possible to "Clone" MKS Cura Plugin? It can do preview pictures also.

oldman4U, 2020-03-13:
@guruathwal 

Sorry to bother you. Would you or another "senior" engineer so kind to have a look at this ticket. Could be a Marlin issue, eventually.

Thank you

https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/103

guruathwal, 2020-03-14:
@oldman4U I don not even know the E of an engineer. I graduated in Finance Major :)
To me, it looks like a Marlin related issue if there is no hardware issue.

guruathwal, 2020-06-03:
Hi @oldman4U 
Need to test some changes to the firmware. can you help?

oldman4U, 2020-06-03:
Hi Guru. It would be a pleasure for me to help you. I have a TFT28 and a
TFT35 E3 v3. Please let me know how i can help you.

Georg

guruathwal, 2020-06-03:
Great.. can we continue through email? My email is on my GitHub profile.

oldman4U, 2020-06-03:
Ahhh. Top secret. 😁I will contact you in about one hour.