bigtreetech/BIGTREETECH-TouchScreenFirmware #240: [Question]TFT 35 v3.0 touch mode endstops don't work

Keltere, 2019-12-15:
I did connect my tft35 v3.0 to my skr pro 1.1 but in touch mode each endstop won't work, if i use the marlin one without the touch all endstops will work as usual.
Can't understand what's the problem.

PGrave, 2019-12-17:
As the TFT mode sends G1 (which should actually be G0, as recommended for moves without extrusion?) commands to move axis, that code is not aware of the current position or even if endstops (max, min, software) are actuated or not. The default operation should be aware of this and give only valid commands. This means that the TFT should know the bed size, thus knowing max absolute positions, and also how close to min and max limits. With a smart coding, no illegal commands would be issued. For instance, before homing, Marlin wouldn't accept any negative movements. After homing it will only move to max positive value.

Keltere, 2019-12-17:
@PGrave  Ok it appears it need a full home, if i try to just home x or y it won't work and if i will give them a negative movement it will crash. Don't know if it's a Marlin fault or TFT35. Thanks btw.

Keltere, 2019-12-17:
It's related to marlin, i did test it with usb and same thing happened.

closed 2019-12-17 by Keltere