bigtreetech/BIGTREETECH-TouchScreenFirmware #241: TFT24 V1.1 Serial connection to SKR Pro 1.1

Lyr3x, 2019-12-15:
I just got my SKR Pro v1.1 and a TFT 2.4 V1.1 which are working fine, except the serial connection between the TFT and the SKR. The Marlin mode is working without any problems however the Touch Mode is only showing no printer attached. 

SKR and Display Baudrate is at: `250000`

SKR Pro runs the latest Marlin 2.0 with the following port configuration:
```
#define SERIAL_PORT -1 
#define SERIAL_PORT_2 1
#define BAUDRATE 250000
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1
#endif
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
```

Any ideas?

Evovita, 2020-01-29:
Try setting the BAUDRATE to 115200 (this is what I use on a TFT3.5 connected to SKR Pro 1.1). Also check that your TFT ribbon cable is making a good physical connection at either end - I had "No printer attached" errors and it was caused by a faulty connector on the ribbon cable that came with the TFT.

JimGWIn, 2020-01-29:
Try setting the BAUDRATE to match your skr pro. mine is 250000 so when i first update it does not work until i make the change

Evovita, 2020-01-30:
@JimGWIn Can you confirm that the TFT runs successfully, including printing, when connected to the SKR Pro 1.1. at 250000 baud? I was led to believe that the current range of BigTreeTech TFTs (inc. TFT24 and TFT35) won't work well above 115200 (which is the default setting in the TFT firmware).

JimGWIn, 2020-01-30:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/241?email_source=notifications\u0026email_token=AMPGZMHEHSMON2VEF6BG3MTRAIQEVA5CNFSM4J27Z6YKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKJJZ5A#issuecomment-580033780",
> "url": "https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/241?email_source=notifications\u0026email_token=AMPGZMHEHSMON2VEF6BG3MTRAIQEVA5CNFSM4J27Z6YKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKJJZ5A#issuecomment-580033780",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

This is what I get at any speed but it all seems to function. Sent from my Samsung Galaxy smartphone.

JimGWIn, 2020-01-30:
Jus for the info, the out of the box pro 1.1 did not function for me. I had to compile and install marlin via the pro micro sd and Bob's your uncle, it all worked. Sent from my Samsung Galaxy smartphone.

[
{

closed 2020-08-17 by bigtreetech