bigtreetech/BIGTREETECH-TouchScreenFirmware #243: Gcode viewer before print photo

manarsak, 2019-12-17:
Hello, why this feature is not present in the screens, and displaying an image of the form before printing will be great

[image]

oldman4U, 2019-12-19:
Hello manarsak.

There is some "model preview" related code in the BTT TouchScreenfirmware, but i never saw the preview on screen;-(

**/**
 * Enable gocde files list mode
 * It is friendly to display long file name, but the model preview feature is not available
 * Disable this if you want to use the model preview feature
 */
//#define GCODE_LIST_MODE**

Maybe you can find out if this is related and if yes, how it works - and post it here.

manarsak, 2019-12-19:
Thank you, but there are screens that support this feature such as mkstft35 or prusa mini The screen displays the form

oldman4U, 2019-12-19:
Yes, i know and it would be great to get the same functionality also for the TFTs from BTT. The code above is from the BTT TFT firmware for Marlin, so there is a way to get this done or i misunderstand the term "model preview".

Hukuma1, 2019-12-20:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/118

manarsak, 2019-12-29:
> It's mentioned how to do here: #118
> 
> Plus suggested for other screens as well. (printing/right before printing)

Can you help me how can I activate this feature

Hukuma1, 2019-12-30:
Which feature? The icon of model like here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/118#issuecomment-552376247

That is simply you have a BMP file with the same name as your model on your SD card and it will load the image automatically. But you must generate this image manually, which to me, seems quite cumbersome.

Hukuma1, 2019-12-31:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/157#issuecomment-569997200

To get the icon of model to show up, just name your BMP file to the name of your model file.

benchy.gcode
benchy_95.bmp (if using TFT 35 the size of icons is 95x95)

blackus3r, 2020-01-14:
It would be nice to have a solution which works with the MKS cura plugin. This plugin creates a small image and paste it into the gcode. The MKS Touchscreen firmware extracts the image from the gcode. 
Or another solution could be a new cura plugin, which will create a .bmp file while saving the gcode on sd card.

bagobor, 2020-02-26:
@blackus3r  FYI, MSK code to read image from gcode file is here: https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware/blob/8e945354770518ebfafcc60f4b952c987cf41d48/User/ui/draw_ui.cpp#L1828

closed 2020-08-14 by bigtreetech