bigtreetech/BIGTREETECH-TouchScreenFirmware #244: TFT35 V3.0 Stop printing random when printing from TFT35 SD card

billydragon, 2019-12-18:
I have BTT SKR v1.3 and TFT35 V3.0. Every things fine. I can printting in LCD simulator mode with no problem. My Onboard card i keep as default to update firmware only and Marlin can't access it. Marlin not see this card. I only use SD card from TFT35 board. Problem is when i printing some file from TFT35 in Touch Mode, printer random stop, i  don't know  why. On LCD emulator mode printing with no problem. 
Any body have an idea to solve this?
* Serial ports setting at 250000 speed.

m-kozlowski, 2019-12-18:
Try with host keepalive disabled (M113 S0).
I'm having same issue and currently testing this workaround. So far it seems to yeld good results.

billydragon, 2019-12-19:
I has disabled keepalive from start of config but still lose communication random. I will try disable all not in use UARTs of TFT35 (like Wifi, UART3,4) because may be conflig on that ports.

billydragon, 2019-12-19:
Yes. Some thing conflig with UARTs in TFT35 V3.0.
in variants.h:

#if defined(TFT35_V1_0) || defined(TFT35_V1_1) || defined(TFT35_V1_2) || defined(TFT28_V1_0) || defined(TFT24_V1_1)
  #define SERIAL_PORT   _USART1  //default usart port
#elif defined(TFT35_V2_0)
  #define SERIAL_PORT   _USART1  //default usart port
  #define SERIAL_PORT_2 _USART2
  #define SERIAL_PORT_3 _USART3
#elif defined(TFT35_V3_0)
  #define SERIAL_PORT   _USART2  //default usart port
 // #define SERIAL_PORT_2 _USART1     //I Disabled all don't use UART
//  #define SERIAL_PORT_3 _USART3
//  #define SERIAL_PORT_4 _UART4
#endif

Now I can printing from TFT Touch Mode without random stop.

Hukuma1, 2019-12-21:
Can you try to print same thing off USB? Wondering if it's SD issue possibly.

billydragon, 2019-12-21:
I try without USB connection, try disabled all not in use UART 2,3,4 in variants.h but still printing from touch Mode randomly stopped. In LCD12864 emulator Mode no problem. I think some interrupt happened when TFT Touch mode print via serial port to Marlin. This is very confused when spend big money for TFT and can only use like normal 12864 LCD.

Sewyboowy, 2020-02-23:
I am having the same issue. Did you manage to fix it? I tried updating firmware to the latest yesterday and still having the same problem.

MrAntonka, 2020-02-23:
Hi

I have exactly the same problem. randomly stops printing.
BTT TFT E3 3.0 ( lastest FW grom git, prepared images for SE card) 
Marlin 2.0-bugfix ( cloned a few hours ago)
BTT SKR1.4 Turbo

I am not able to print from the SD card.

Sewyboowy, 2020-02-23:
Yeah I updated marlin and screen firmware yesterday. Tried with baud rates 115200 and 256000. I've got reprap discount smart controller disabled in marlin but the EXP connectors plugged in anyway, wondering if that's got something to do with it. Seems to print fine connected directly to a computer

Doenermaker, 2020-02-23:
Same problem here. SKR1.3 and TFT35v1.2. Latest Marlin 2.0.4 from today. Tried different tft firmwares (standard, Guruathwals fork pre compiled 1.24.2 and 1.25.2 and self compiled 1.25.2). No difference.

Randomly stops printing, sometimes on layer one, sometimes much later, even on the same GCode-file. Printer needs to be reset to accept TFTs signal again.

Sewyboowy, 2020-02-23:
Well I'm glad I'm not alone.
Any ideas how to get BTT to help?

Hukuma1, 2020-02-23:
Just to confirm it's not a Marlin change, can you guys try Marlin 2.03?

https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.3

Scope666, 2020-02-23:
There's been major development activity lately, you should go NON-BUGFIX, which the latest release is now 2.0.4.2.   It's actually ahead of Bugfix right now which is strange.

https://github.com/MarlinFirmware/Marlin/releases

Sewyboowy, 2020-02-23:
Brilliant thanks I'll give this a try tomorrow

MrAntonka, 2020-02-24:
Hi, I have re-flash TFT again and now it works perfectly :) do NOT use precompiled SD images. Compile your own bin file.

Now it works well :)
Roman

Sewyboowy, 2020-02-24:
Hmmm I've been compiling my own as well but having the same problem, what versions of marlin and TFT firmware are you using?

MrAntonka, 2020-02-24:
Marlin-2.0.x-bugfix - commit : 2bcc2ec7d1125428beb6898a527d1cc49c26ac78
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware - commit: 5777f41b5f8c1a41410a6874614499c91ac78fa2

all latest

Sewyboowy, 2020-02-24:
Just tried with marlin 2.0.4.3 (which I believe has the fix from 2bcc2ec7d1125428beb6898a527d1cc49c26ac78) and I have the same touchscreen firmware.
The print still fails, but now when it does the i'm taken to the home screen and everything remains responsive. I also then tried to re print the same part and when clicking "print", I got a message asking if I'd like to resume the print.
Any ideas? How do you have your tft configured in marlin and have you done anything special with the tft firmware?

MrAntonka, 2020-02-24:
@Sewyboowy  do you have EXP1 + EXP2 connected?

Sewyboowy, 2020-02-24:
@MrAntonka Yes I have both connected and the TFT cable. I've got the "full graphic' display enabled in marlin but not the "smart controller"

MrAntonka, 2020-02-25:
hmm,, I have no idea then

Sewyboowy, 2020-02-25:
@MrAntonka would you mind sharing your configuration and configuration.adv files so I can see where we differ?

MrAntonka, 2020-02-26:
yes np ....  give me your email :)

Sewyboowy, 2020-02-26:
Thanks! My email is
tom-sewart@hotmail.co.uk

MrAntonka, 2020-02-26:
check your mailbox ;)

Sewyboowy, 2020-02-26:
Got it thanks!

piotrekm7, 2020-02-27:
I've flashed latest marlin and touch screen firmware and the problem still persists.
@MrAntonka can you share your configs publicly ?
@Sewyboowy any luck with that ?

Sewyboowy, 2020-03-02:
Sorry I've been so slow, been very busy!
I've just looked at MrAntonka's config, the only notable differences being that in mine I had ST7920_FULLSCREEN and UNIFIED_MENU commented out. I've uncommented them now and running a test print.

Sewyboowy, 2020-03-02:
Neither of those options fixed it

Sewyboowy, 2020-03-02:
Am having a look at marlin config now but cant test until tomorrow. Here are MrAntonka's config files.

[marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4277114/marlin.zip)

[TFT.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4277115/TFT.zip)

MrAntonka, 2020-03-02:
When you switch to classic display simulator, does it works?

Sewyboowy, 2020-03-02:
I tried that on an older release of marlin (maybe a month or so ago) and it failed

MrAntonka, 2020-03-02:
When you switch to "LCD12864 simulator" are you able to print?

Sewyboowy, 2020-03-04:
Previously printing with the LCD12864 simulator had the same issue, but just printed a benchy fine. Any ideas how to fix touch mode then?

Sewyboowy, 2020-03-04:
Just tried another print in LCD12864 sim mode and it failed. Guess it doesn't work

reQu1em00, 2020-03-16:
I have the same issue.  Using an SKR mini E3 1.2 with the tft35 on an ender 3 pro.  Random drop outs when I print from anything in the tft.  Also  the tft will freeze sometimes if I print from octoprint but the print continues.  It usually says ‘processing’ in red at the top of the screen.

Sewyboowy, 2020-03-21:
Mine works in sim mode now that I have enabled reprap smart controller. Still got the same issue otherwise. People have suggested insulating the cables from EMI but I'm not sure... I get the same issue even if I am printing from SD in sim mode but connected to repetier as well. Really odd

kingontheroad, 2020-04-04:
Same here today after 2,5 hour printing , pats drop out grrr

Has someone of you , solved the problem ?

piotrekm7, 2020-04-04:
@kingontheroad Yeah, I have bought different display. I tried dozens of configurations and nothing helped, even making new heavy duty connection cable.

kingontheroad, 2020-04-04:
> 
> 
> @kingontheroad Yeah, I have bought different display. I tried dozens of configurations and nothing helped, even making new heavy duty connection cable.

May i ask , what for display you bought 
Iam thinking about the same to do and icm other motherbord

piotrekm7, 2020-04-04:
@kingontheroad Classic "reprap discount full graphic smart controller". I much prefer fancy TFT touch display, but it's useless for me with this bug.

kingontheroad, 2020-04-04:
> 
> 
> @kingontheroad Classic "reprap discount full graphic smart controller". I much prefer fancy TFT touch display, but it's useless for me with this bug.

Iam gonna sell my btt board + tft and buy mks 32bits board with mks screen .
I have this set from btt now 4 weeks and 4 busy weeks busy with pionering and fixes bugs...

Thanks for the reply @piotrekm7

billydragon, 2020-04-04:
Yes. I spent many much money buy BTT product but they are bad quality and not finish developed. SKR Pro v1.1 after some months work now it never booting up. TFT35 V3 as you see problem here, closed loop motor never work normally after buy it. So, solution is change to makerbase MKS products and not headaches about bug.

kiwipaulrob, 2020-04-13:
My problem turned out to be a problem with the micro sd card I was using. Check your card.

phlebaspho, 2020-05-26:
In my case, after much A/B testing I realised it was not the TFT35 - it was "device connected to Marlin serial port 2" that experienced the the comms loss. Moving the TFT35 to the frist serial port in Marlin config.h fixed it.

piotrekm7, 2020-05-29:
I managed to have it working as well. In my cause, reducing number of TMC microsteps from 256 to 32 did the job.

mderdouri, 2020-06-19:
> I managed to have it working as well. In my case, reducing number of TMC microsteps from 256 to 32 did the job.

Can you please post your config files publicly for reference? I've been dealing with this bug for weeks and can't get it fixed.

merinkumar, 2020-07-10:
> In my case, after much A/B testing I realised it was not the TFT35 - it was "device connected to Marlin serial port 2" that experienced the the comms loss. Moving the TFT35 to the frist serial port in Marlin config.h fixed it.

Can you please let know how to do that?

merinkumar, 2020-07-11:
I Guess this is what we need to do for changing the port.
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

here 0 is actual UART0 to which the TFT is connected and -1 is the virtual USB for PC communication. Earlier I had the -1 and 0 other way , and the print never takes off after G28, it just hold the position and not even stop cmd works. Now seems working.

Gordon44, 2020-07-20:
If I set #define SERIAL_PORT 0
I get this error message any Ideas what is wrong 
The terminal process "C:\Users\gegro\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'BIGTREE_GTR_V1_0'" terminated with exit code: 1.
It will work if I set #define SERIAL_PORT -1
The other #define SERIAL_PORT 2-0 seams to have no effect

ovladaf, 2020-08-07:
Here the same problem... I'm mad with this tft35...

DennisHeijmans, 2020-09-13:
Recently i compiled marlin 2.0.6.1 for bij E3 DIP V1 and set the first serial port to -1 and the second to 2.
Als updated TFT35 firmware to the latest.
I can even use the onboard SD from the SKR board in touch mode now.
Anyway had no issues printing from TFT SD so far. Maybe i solved it ?

DennisHeijmans, 2020-09-18:
That was not the Solution but maybe the latest TFT35 firmware with some Different settings on the SKR is. Serial and serial2 need to be 2 and -1

DennisHeijmans, 2020-09-25:
I found this and changed some more settings in Marlin : https://www.makenprint.uk/3d-printing/3d-printing-guides/skr-e3-dip-marlin-2-setup-part-1/

Also compiled the TFT35 myself this time.

MarcBernat, 2020-09-28:
Same problem here. Print from Onboard SDCard, bed/nozzle warm-up, G28, Z moves 2 mm, stuck in position.

TFT34 E3 V3.0.26.x (12 Sept) with Marlin 2.0.6.1 release (Sep 27)

-- Ender 3 Custom Start G-code --
G92 E0 ; Reset Extruder
G28 ; Home all axes
G1 Z5 F5000  ; 
M75 ; 
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

DennisHeijmans, 2020-09-28:
I have complex New mArlin firmware and changed some buffer settings. Also tft firmware is now compiled by me. Done a 20 hours print and 4 hours print with no issues from SD in the screen in Touch mode.

MarcBernat, 2020-09-28:
Found a Workaround:

**Remove M75 command from your Slicer start G-Code**

M75 - Start Print Job Timer  [https://marlinfw.org/docs/gcode/M075.html](url)

After comment this issue I was thinking that the "printer screen mode" was already initiated the timer count system. So, maybe the issue was execute M75 again (unnecessary)

DennisHeijmans, 2020-09-28:
> Same problem here. Print from Onboard SDCard, bed/nozzle warm-up, G28, Z moves 2 mm, stuck in position.
>
> TFT34 E3 V3.0.26.x (12 Sept) with Marlin 2.0.6.1 release (Sep 27)
>
> -- Ender 3 Custom Start [...]

The problem I have is only when printing from TFT35 SD card and in Touch mode.

oldman4U, 2020-10-09:
@billydragon Are you still watching?

What is the status of the reported issue for you?

DennisHeijmans, 2020-10-09:
After compiling the tft firmware myself from the latest release and Changing the Serial buffer settings I have not had any issues with this, even done a 24h print flawlessly.

oldman4U, 2020-10-10:
Which serial buffer settings and which values?

[28 more replies]

closed 2021-04-07 by stale[bot]