bigtreetech/BIGTREETECH-TouchScreenFirmware #247: popup status window

andrzej1974, 2019-12-20:
Hello
I have question. How change or disable popup status screen  in during printing popup. Because every layer this windows show information about with layer now is printing and cover main printing menu. Also we have this status screen in main menu ( not a popup ) before start print, but if we start printing, menu is change and still we have popup status screen.

guruathwal, 2019-12-21:
you have to change setting in your slicing software to stop the messages on layer change.

andrzej1974, 2019-12-21:
I'm using Cura. Do you know where that can change?

PGrave, 2019-12-30:
@andrzej1974 Check your Gcode files for M117 and remove. This is the Set LCD command, which forces a popup on the TFT. If you want a more permanent solution, edit your slicer preferences. In Cura there's a setting in (Menu)-Extensions-Post Processing-Modify G-Code. If script "Display Layer and Filename on LCD" is active and enabled, deactivate.

andrzej1974, 2019-12-30:
Thanx. This was help.

closed 2019-12-30 by andrzej1974