bigtreetech/BIGTREETECH-TouchScreenFirmware #248: [BUG] Overlapping flash storage address of icons with Fonts❗

guruathwal, 2019-12-20:
@bigtreetech 
according to the following line 
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/703de8f83eb4c181564bc5f286afa9824cce11cb/TFT/src/User/API/boot.h#L11
the address of the last icons (No.108 to be exact) which will be `0x573000` ‬goes beyond the stating address specified here:  https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/703de8f83eb4c181564bc5f286afa9824cce11cb/TFT/src/User/API/boot.h#L14

it looks like the addresses specified in above lines are for 80 icons and the `word_unicode.fon` gets overwritten by rest 28 icons (819200‬ bytes total). this might result in issues with some languages.

@bigtreetech  @Msq001 can you please confirm it if this is the case?

HappyCodingRobot, 2020-01-02:
No new information on this?

extesy, 2020-01-05:
@bigtreetech @Msq001 ping...

gilliands, 2020-01-06:
Any progress on this?  I had a go at fixing it myself using this cack-handed code that supports up to 128 icons before overlap.  I should add some range checking though to check for overflowing the end of EEPROM.

//address in spiflash W25Qxx
#define LOGO_ADDR               0x0
#define LOGO_length             0x4B000 //0x71000
#define ICON_ADDR(num)          ((num)*0x5000+LOGO_length)
#define WORD_UNICODE            (128*0x5000+LOGO_length)
#define WORD_UNICODE_length     0x480000
#define BYTE_ASCII_ADDR         (WORD_UNICODE+WORD_UNICODE_length) // ascii (+0x1000 4K) //0x700000
//#define BYTE_RESERVE_ADDR       (BYTE_ASCII_ADDR+0x1000)

#define SMALL_ICON_START_ADDR   ICON_ADDR(ICON_PREVIEW+1)
#define SMALL_ICON_ADDR(num)    ((num)*0x1000+SMALL_ICON_START_ADDR)

closed 2020-05-04 by guruathwal