bigtreetech/BIGTREETECH-TouchScreenFirmware #249: [BUG] Firmwareupdate stuck at 1% on TFT24 when connected to SKR 1.1 pro EXP2

arminth, 2019-12-20:
I just switched from SKR 1.3 to SKR 1.1 Pro.

I realized that when you have an SD inserted with Firmware, after a poweron or reset, the update starts but Dsisplay gets stuck at 1% for a moment and then display goes black.
Removing the EXP2-cable let the upgrade immediately continue (when stuck at 1%) or let it run correctly  after a reset.
SKR 1.3 doesn't have that issue.
Weird!

@bigtreetech PLease fix or advice.
Cheers
Armin

guruathwal, 2019-12-20:
can you try it by removing just the reset pin from the cable and see what happens?

arminth, 2019-12-20:
That was the first thing I tried. Didn't help. I had removed the tft cable entirely and just left exp 1 and 2. Still the same.

guruathwal, 2019-12-21:
There is a reset pin on EXP1 also:

[image]

arminth, 2019-12-22:
I don't want to cut my exp1-cable honestly....
It obiously has something to to with signals on exp2. The moment I plug it in, anything SD-carsd related during Update stops. Unplugging exp 2 and it continues.
I think somebody from @bigtreetech should have a look!

arminth, 2020-03-31:
@bigtreetech Update!?

oldman4U, 2020-10-09:
Is this with the same TFT24?

arminth, 2020-10-21:
Yepp

oldman4U, 2020-10-21:
Reading your 2 tickets a few times, I get more and more the feeling that the SKR Pro has an issue: I would contact the dealer for a replacement.

oldman4U, 2020-10-21:
Do you want to give me access to your Pro firmware vie WeTransfer or something similar?

arminth, 2020-10-22:
https://github.com/arminth/Marlin/tree/Kossel_neu_SKR1.1Pro

oldman4U, 2020-10-22:
OK. Downloaded your files, compiled them without a single error or warning;-) uploaded it it my SKR Pro with the TFT35 E3 v3 connected and I can select a gCode file on the SD card to print without a problem.

Hope this helps

[arminth.mov.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5421876/arminth.mov.zip)

arminth, 2020-10-22:
Thanks for the help!
I also tried to dig deeper and used some dupont wires as in between connectors to trace down, which signals cause the issues.
I found that it's the SS (aka CS) and the SW (aka DET ). If one of these is connected to the 2.4" display, SD is not readable in touchmode. You even have to reset after disconnect to make it work again. MAybe it's apullup-pulldown thing. I don't know. I was about to order a 3.5" but it's not available in the czech warehouse so no shortterm test. And the urge is not soooo bad to ordr a new SKR 1.2 Pro... Spent enough money on that Kossel already.

oldman4U, 2020-10-22:
Your very welcome. In the meantime I have several printers worth in spare parts for testing. Strange hobby😁

Please do not forget to close the tickets once you verified the hardware issue.

Thank you and happy printing

closed 2020-12-28 by stale[bot]