bigtreetech/BIGTREETECH-TouchScreenFirmware #252: Print information on tft mode from onboard sd files.

andrzej1974, 2019-12-21:
Hello
If i start print from onboard sd files in tft mode i dont recive any information about status, like temperature from bed or hotend. There is possiblity add this information to the menu?

bugattko, 2019-12-22:
same problem, sometimes there is no update status information, which settings in marlin are neccesary for communication with tft ?

Piscanc, 2019-12-22:
Hi 
I am not on very latest versions of FW, but close... i can describe behavior of screens, since i have both connected.
printer: modified ender 3
board: e3 mini
screens: stock ender3 + tft35 v2

both screens power up as expected, tft reports printer ready as soon as lcd finishes power up sequence.temp ok, after homing both screens report correct positions. 
now the "funny" part:
depending from where you start print from, that screen will show status from then on.
if you start print from ftf, lcd will remain at printer ready status, but temp and position will update. all the settings are seemingly accessible, but i figure executed after previous job is finished, depending on job type. heating being one of the most problematic ones...

if connected over usb:
send pid auto-tune command, only lcd will report temp change during progress. tft status remains on temp at witch it was started on, seemingly freezing but will refresh after auto-tune cycle does it´s job.
not related to screens, but i could not get bed pid to work.

and another thing to test: reset after error (temp runaway)
trigger temp error, both screens will display error, press reset on tft.
only tft will reset, board will not, lcd still shows error.

will test latest versions in few days,  as soon as i finish latest set of mods for printer.

Piscanc, 2019-12-27:
now tested with vanilla marlin 2.0.1 and latest touchscreen FW...
still same with print status info, it´s shown depending from start location.
lcd now has display overlay issues with text from printing file info and temp.
on top of that i now get dot artifacts on prints, i presume from pauses on information refreshes on display, since dots correlates with blinking light in top right corner of lcd.

oldman4U, 2020-10-22:
The temperature is shown using Marlin 2.0.7.2. please check and close the ticket afterwards. 

Thank you

closed 2020-12-28 by stale[bot]