bigtreetech/BIGTREETECH-TouchScreenFirmware #254: [BUG] TFT24 v1.1 (White Screen)

JevonW, 2019-12-22:
### Description

TFT goes white after powered on for a minute or two, when it's white it will remain white if you reset the board. If left unplugged for a while it will start normally in both simulation or LCD mode, only to return to a whit screen after some time has passed.

This issue started after deploying 1.1.25.

I have tried the archive firmware (BIGTREE_TFT24_V1.1.24.2.bin) the current firmware (BIGTREE_TFT24_V1.1.25.bin) both variants (Standard and Unified) with no success, I also compiled from the latest source and the symptom persists.

### Steps to reproduce

1. Upload BIGTREE_TFT24_V1.1.25.bin to SD card and include the TFT24 folder containing the BMP
2. Insert SD card into TFT24 module
3. Restart / Power on TFT24 module, wait for firmware update to apply

**Expected behavior**
Working TFT module

**Actual behavior**
TFT sometimes works and when it does it will display a white screen and become non-responsive

Drebbs, 2019-12-25:
Hi, I have the same issue.

I've copied Folder and File (BIGTREE_TFT24_V1.1.25.bin) etc to SDcard. I don't get any message about an update, just a white sceen.

Now all I have is a white screen! sometimes a calibration window, but then a white screen.

JevonW, 2019-12-26:
[image]
[image]

I managed to upload the latest firmware and after a couple of minutes (+- 3) it went white again... I'm beginning to think it might be a hardware fault which really sucks considering it's brand new and the return shipping to China will cost more than a replacement unit.

Perhaps a faulty capacitor or something that over-currents the screen?

JevonW, 2019-12-26:
Video of the fault:

https://drive.google.com/open?id=1R9Rcb5AyhSI7BMfuwrinOqdfOL-nED1r

JevonW, 2019-12-26:
> 
> 
> Hi, I have the same issue.
> 
> I've copied Folder and File (BIGTREE_TFT24_V1.1.25.bin) etc to SDcard. I don't get any message about an update, just a white sceen.
> 
> Now all I have is a white screen! sometimes a calibration window, but then a white screen.

Try leaving it unplugged for a while this seems to help bring it back to life for a minute or two, I suspect it's a capacitor or something that's supplying too much voltage to the LCD

nagimitsurugi, 2020-01-19:
Hello everyone.
I have the same issue.
I was having trouble with the exact same behavior as the video above.

The board was still operational after the screen went white, so I suspected that the TFT was defective.
So, I got a TFT panel that works with the same controller chip that I found on Ali Express and replaced it, and now I can use it without any problems.
After all it seems that the LCD panel is defective.

Each item costs about $ 5, so it might be a good idea to try it.

randyabm, 2020-01-24:
> Hello everyone.
> I have the same issue.
> I was having trouble with the exact same behavior as the video above.
>
> The board was still operational after the screen went white, so I suspected that the TFT [...]

Hi nagimitsurugi Would you mind to share the TFT that  you got.

nagimitsurugi, 2020-01-24:
> Hi @nagimitsurugi Would you mind to share the TFT that you got?

Speaking of which, I should have put a link. sorry.
I tried this product.
https://aliexpress.com/item/32837085500.html?spm=a2g0s.9042311.0.0.28b04c4dCjE0rd
The product is selected by specifying the color, but it is specified by "2.4 Touch" in it.
I bought two of them because they had to be bad, but both were usable without any problems.

It seems to be a part used for the touch panel shield used for Arduino UNO and so on.
When I looked at the shield I had, it was the same.

I don't speak much English, so I use translation software. I'm sorry if it's hard to understand.

DJDOUBLEG, 2020-05-06:
https://www.youtube.com/channel/UCRr_ZiDkVxE6sd-FbRgmrQw many users have same issue.
The solution is replace the TFT screen module.

ChritstianAlexander, 2020-08-09:
I have the same problem. I warm it up with a hair dryer for 30 sec and it is fine. The problem is that it cools down again and stops the printer mid print. I too think it is a hardware problem. I have V2.0.

DJDOUBLEG, 2020-08-09:
I make waranty claim. And BTT send to me new tft module.

ChritstianAlexander <notifications@github.com> ezt írta (időpont: 2020.
aug. 9., V 23:01):

closed 2020-08-14 by bigtreetech

shahzaibkamboh7, 2022-07-01:
I have the same issue

briefer666, 2023-04-03:
same problem here