bigtreetech/BIGTREETECH-TouchScreenFirmware #257: [BUG] Z offset wrong calculation/display

Hans007a, 2019-12-23:
# Description
Display TFT35 V3.0, SKR Pro v1.1 and latest marlin bugfix 2.0.x
Z-Offset is set to -1.120, but it shows in the LCD Mode .12
If I save this, my nozzle is more than 1mm above the bed.
In the Simulator is the Value ok, but it comes directly from the Board

### Steps to reproduce

1. Set Z-Offset in Simulator Mode and save to EEPROM
2. Switch to LCD Mode and the Value is wrong
3. Set the Value in LCD Mode
4. Switch to the Simulator Mode and the Value was saved wrong

**Expected behavior**
in LCD Mode a correct Z-Offset Value

**Actual behavior**
in LCD Mode wrong Z-Offset Value

### Additional Information
none

Regards
Hans

UnknownHero99, 2019-12-23:
what happens if you set it through LCD mode? I don't own a v3 but it works fine on v2 (only LCD mode) 
although the save button probably doesn't work as same for the same reason as issue #253?

also, could you try setting it on simulator mode and saving EEPROM and then check in LCD mode

closed 2019-12-29 by Hans007a

teriksm, 2019-12-31:
The Z-offset value is once again incorrectly represented in V1.1.25 (TFT24), I believe that it was ok on .24.