bigtreetech/BIGTREETECH-TouchScreenFirmware #259: [BUG] INVERT_ZAXIS: sign in gcode swaped

HappyCodingRobot, 2019-12-24:
### Description

Imho, the implementation of the INVERT_ZAXIS is not correct.
In particular the sign in ZGODE_UP & ZGCODE_DOWN.
(see definition of INVERT_YAXIS)

### Steps to reproduce
Move Z axxis in move menu

**Expected behavior**
Up move should send positive moving value

**Actual behavior**
Up sends negative value

### Additional Information

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/b90cc1533cd93ead56e7e7cf257ad13d56afae21/TFT/src/User/Menu/Move.c#L24-L34

guruathwal, 2019-12-24:
it looks like the gcodes need to be swapped..

HappyCodingRobot, 2019-12-25:
@guruathwal yes, I agree with that.

HappyCodingRobot, 2019-12-26:
@guruathwal I'm not sure if I fully understand your changes atm. Ok, the defines now only work in non-list-mode, otherwise it is set via configuration menu. Nice change.. 
But I'm afraid, something got lost during merge:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/2a6211e87c8370a8b29394132d22158bca02820b/TFT/src/User/Menu/Move.c#L33-L34

Also the inverting in list-mode doesn't seem to work here as intended. Didn't found yet why ..

guruathwal, 2019-12-26:
@HappyCodingRobot thats because the Minus sign is missing in line number 34.. 😂

HappyCodingRobot, 2019-12-26:
yes .. ;)
but actually I can't find find the cause of the issue in list-mode.
btw, thanks for your great work :)

guruathwal, 2019-12-26:
> yes .. ;)
> but actually I can't find find the cause of the issue in list-mode.
> [edit] issue: ICON_Z_INC always has Z- label and moves down, _DEC has Z+ moves up regardless of the settings
> 
> btw, thanks for your great work :)

@HappyCodingRobot 
👍
Fixed it now.. #263  I think...

HappyCodingRobot, 2019-12-26:
@guruathwal thx, it works great.
Just one minor detail I saw; Icon_0 is the decrease icon and should send a GCODE_DEC in non-inverting mode/setting and Icon_2 vice versa imho

guruathwal, 2019-12-27:
@HappyCodingRobot 
The z movement icons are a bit confusing.
The meaning of the icon is different for different printer types.
For printer with bed moving up down the icon_0 is viewed as inc_z while in printers with x axis moving up down it is viewed as dec_z.

HappyCodingRobot, 2019-12-27:
@guruathwal ok, get your point. I already branched for my specific changes, so no problem.

HappyCodingRobot, 2019-12-27:
Closing, fixed with #263

closed 2019-12-27 by HappyCodingRobot

ipatch, 2019-12-27:
@guruathwal @HappyCodingRobot 

not sure if this is the desired behavior or not but i just pulled in all upstream changes from this repo to my local fork, compiled with vs code, and copied the unified menu icons for my particular screen, ie. a _BTT TFT35v2_

and this is what i'm seeing on the move menu, the **Z+** text is below the ⬇️ _down arrow_ icon.

_Is the intended behavior because the ⬇️ is moving the gantry up on my particular printer?_

##

[image]

guruathwal, 2019-12-27:
@ipatch goto to setting>feature setting> invert z axis

ipatch, 2019-12-27:
@guruathwal 

is that the same a defining it in the `Configuration.h` file?

ipatch, 2019-12-27:
@guruathwal 

I believe the text labels are correct after inverting the z axis however the icons appear to be in the wrong spot, no?

HappyCodingRobot, 2019-12-27:
@ipatch like @guruathwal above mentioned, it depends on the printer. For e.g. a moving bed this would be ok as for larger z values, the bed moves down.
For my machine i decided to make a separate branch in my fork to match the build.