bigtreetech/BIGTREETECH-TouchScreenFirmware #261: Add access to cd tft onboard from ESP3D

Alex-TV, 2019-12-25:
Ability to manage tft onboard cd from ESP3D.
Copy files to cd tft onboard .
Start printing from cd tft onboard .
Add #define to Configuration.h. For the ability to switch access to cd tft onboard  or cd printer

luc-github, 2020-01-04:
@Alex-TV  I have opened tickets on my repo https://github.com/luc-github/ESP3D/issues/393, https://github.com/luc-github/ESP3D-WEBUI/issues/83
I hope be able to work on this soon

Alex-TV, 2020-01-04:
> @Alex-TV I have opened tickets on my repo [luc-github/ESP3D#393](https://github.com/luc-github/ESP3D/issues/393), [luc-github/ESP3D-WEBUI#83](https://github.com/luc-github/ESP3D-WEBUI/issues/83)
> I hope be able to work on this soon

Thank you very much, I will look forward to it. Can I help you with something?

luc-github, 2020-01-04:
I just need to find some time - I have bought  a BTT TFT and ESP01S so I need to study tft code and do some tests - I have some actions on other projects to finish before working on this topic
 I will update the ticket when there will have something to test so you can have try ^_^

luc-github, 2020-04-20:
I have implemented the basic GCODE commands  for listing and printing from TFT SDBoard and TFT USB disk - I will add the support on WebUI later
please check -https://github.com/luc-github/ESP3D/issues/393#issuecomment-616444547
and https://github.com/luc-github/BIGTREETECH-TouchScreenFirmware/tree/esp3d-support

Feedback are welcome - once validated I will push a PR

luc-github, 2020-04-26:
Update :
It is now implemented it in 2.1-BTT-TFT
Need https://github.com/luc-github/BIGTREETECH-TouchScreenFirmware/tree/esp3d-support
and https://github.com/luc-github/ESP3D-WEBUI/tree/2.1-BTT-TFT
and of course latest 2.1 ESPD3D FW https://github.com/luc-github/ESP3D/tree/2.1

In preferences there are some settings to add TFT SD and TFT USB support in main UI
[image]

Which give:

[image]

[image]

[image]

Upload is not functional at all for TFT SD neiter TFT USB, because there is not support in TFT FW for this  so button is hidden
Upload is currently not Functional due to polling polution - so I left button but need TFT FW update to stop polling when upload is ongoing according M28/M29 commands - TBD later

M23/M24 /M25 / M27 / M27 C / M30 / M524 commands have been implemented in TFT FW

I have change Abort command for Marlin from M112 (emergency stop) to M524 (SD print stop)

Need some test to validate everything is ok and there is not side effect on configuration without TFT and not using Marlin (Repetier / Smoothiware) 

Feedback are welcome , 
Note: rebase has been done to use latest git from BTT and that introduced a problem with images  which is solved by this PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/619 so you may need to to the change manually until PR is merged

luc-github, 2020-05-03:
PR is pushed

closed 2020-08-14 by bigtreetech