bigtreetech/BIGTREETECH-TouchScreenFirmware #267: F0 shows 0 on Unified 3.0.25 TFT35 even it is set to other values via GCODE

broxinagil, 2019-12-27:
My TFT35 shows F0 as 0 on Info-Screen when I print via USB, even the Filament Cooling Fan is running. Other Values, like Hotbed and Hotend temperature are displayed correctly.

Printer: Creality Ender 3 Pro with SKR Mini E3 1.2

broxinagil, 2019-12-27:
As well (Maybe this can be seen as another Issue) the Value on F0 is shown as values from 0 to 255, instead 0-100%

oldman4U, 2019-12-27:
Hello broxinagil.

Have you activated:
#define SHOW_FAN_PERCENTAGE // enable to show fan speed as a percentage instead of a value
in the firmware?

broxinagil, 2019-12-28:
No. I've just installed the bin. Will take a look into sources. Bust this still will not solve the main issue I guess?

guruathwal, 2020-01-31:
@broxinagil  please close the issue if it has been solved.

Taomyn, 2020-04-13:
Mine is the same, compiled version of 3.0.26, the fan says 0% even though it is running - in Marlin mode it shows 100%.

xrunner, 2020-05-01:
Also mine does same (v. 3.0.26 installed yesterday Apr 30, 2020). Always shows 0% (F0 part cooling fan) during prints on main menu (always has). If you go into the fan control menu and control F0 (not while printing), it will then show the percent on the main menu indicator.

kkosa1, 2020-05-14:
Any new insights on this? Same for me since day one currently using V3.0.26.1 (may 8, 2020)

guruathwal, 2020-05-17:
This is a marlin issue as it does not report the current fan speed. If the Fan speed is changed through any method other than the TFT, then the TFT will not know the current fan speed.

I have pushed a PR in Marlin firmware to address the issue: https://github.com/MarlinFirmware/Marlin/pull/18013

PR #733 (2020-07-02): Add parsing of fan speed, Improve popup dialog window, Add large fonts, Improve Numpad UI & other bug fixes
- ~Integrate preheat menu with heat menu.~
- Improve Live icons.
- Improve Numpad UI.
- Use Numpad to set temperature instead of increase decrease button. (Click on heat status icons to edit temperatures.)
- Fix 'T0' temperature updating to the current tool/nozzle.
- Fix bugs in baby-stepping and probe offset menu.
- Improve popup dialog window (display window according to message type).
- Fix unnecessary item re-draw in the feature settings menu.
- Add printf/sprintf library from https: [... cut]
closed 2020-07-02 by bigtreetech