bigtreetech/BIGTREETECH-TouchScreenFirmware #268: [BUG] TFT24 v1.1 Babystepp not going to "-" direction

Daeguhyuk, 2019-12-29:
### Description
Board: SKR v1.3 + TMC2209 UART + TFT24 v1.1
Firmware: marlin 2.0.1 & TFT24 v1.1 .25

If I try baby step, it only proceeds in the + direction.
Trying to move in the-direction cause some vibration in the motor.
It seems hesitant to move in the-direction.
This problem occurs in both TFT mode and 12864 mode.

### Steps to reproduce

1. [First Step]
 - 12864 Mode Try to babystep. rotate knob
 - TFT Mode Try to babystep, push TFT Screen
2. [Second Step]
 - Normal operation in the  '+'direction in all modes
 - Abnormal operation in the '-'direction in all modes
3. [and so on...]

**Expected behavior**

 - Normal operation in the  '-'direction in all modes

**Actual behavior**

- Abnormal operation in the '-'direction in all modes
- Trying to move in the '-'direction cause some vibration in the motor.

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
https://youtu.be/zt0BUbh-B4M

guruathwal, 2020-01-02:
if it is not baby-stepping towards negative values in both modes then may be there is problem in your marlin firmware as in marlin mode the TFT just acts as normal 128x64 lcd display.

Daeguhyuk, 2020-01-03:
> if it is not baby-stepping towards negative values in both modes then may be there is problem in your marlin firmware as in marlin mode the TFT just acts as normal 128x64 lcd display.

I tested my 2004 clcd. It was no issue 
Only TFT24 v1.1 is not good.

Daeguhyuk, 2020-01-10:
I changed moter driver From TMC2209  TO A4988
Babystep works fine.
I cannot understand why TMC 2209 not going - direction, babystepping mode.
Maybe marlin issue..

closed 2020-01-10 by Daeguhyuk

guruathwal, 2020-01-10:
good to know that.. maybe you should open an issue in marlin repository for the same..