bigtreetech/BIGTREETECH-TouchScreenFirmware #273: Position not shown in Unified Menu

dkoenig42, 2019-12-31:
The current position for X, Y and Z is always shown as 0.00 in the unified menu.
Temperatures and Fan speed are updated fine.
Hardware Version is V3.0.

AnHardt, 2019-12-31:
Try to provoke with M114.

dkoenig42, 2019-12-31:
I tried it, Marlin seems to send correct response "X:-22.00 Y:91.52 Z:31.70 E:0.00 Count A:5562 B:9082 Z:57068"
But still no update of the values on the main page.

dkoenig42, 2019-12-31:
Short update: If I send "M114 D", the position is updated once. How do I activate an automatic update?

dkoenig42, 2020-01-01:
I found a solution for me.
I commented out "stepper.report_positions()" in the function report_current_position() in "motion.cpp" in Marlin 2.0. Works fine now.
Looks like the additional stepper information is disturbing the display.

dkoenig42, 2020-01-01:
Sorry to roll back, but my patch leaves me with Octoprint not working any longer.
Taking a look at parseACK.c of the display, it seems to be waiting for a stepper position of the extruder. But current Marlin 2.0 does not output it (any longer?).
So currently I have no solution.

guruathwal, 2020-01-02:
which firmware version are you using? can you share the actual data sent by the printer after M114 command?

dkoenig42, 2020-01-02:
Marlin 2.0.1. 
I made some progress: If I activate "M114_DETAIL" in Marlin, I get updates of the position on the display.
But during printing, the number jump between the correct ones and numbers ways to high. It looks like the display is parsing the move commands from Octoprint and interprets them as positions (maybe because they also contain the strings "X:", "Y:" and "Z.").

guruathwal, 2020-01-02:
@dkoenig42  firmware version of your TFT?

dkoenig42, 2020-01-02:
3.0.25, self compiled with Platform i.o.

guruathwal, 2020-01-02:
can you try the pre-compiled firmware to check if it works. also the position will not update correctly when printing through other host sources as the TFT relies on gcode for position update.

dkoenig42, 2020-01-04:
I tried the precompiled firmware, same result: positions are not updated correctly when printing from OctoPrint. But if I understand you aswer correctly, thats caused by the design of the firmware, right?

guruathwal, 2020-01-31:
@dkoenig42  while printing from octoprint the position cannot be updated correctly as the printer will be printing too fast.

closed 2020-08-14 by bigtreetech