bigtreetech/BIGTREETECH-TouchScreenFirmware #274: [BUG] (Custom Gcode not working in list mode)

gneu42, 2020-01-01:
### Description

### Steps to reproduce

1. [Add custom gcode]
2. [compile]
3. [Test on printer]

**Expected behavior**

**Actual behavior**

### TFT35 V3.0 connected to SKR 1.3

guruathwal, 2020-01-02:
Thanks for notifying the issue... Fixed here.. #279

gneu42, 2020-01-03:
Hi, 
Thank you for this new update, but the custom Gcode still does'nt work. I went back to Non list mode and they are working ok. 
What can I do to get them working ?

While in this mode I had problem with Y movments. I found a bad copy paste in move.c (see below Both Keys are sending YGCODE_UP (lines 163 & 196)
I change key 5 to YGCODE_DOWN and all is ok.

Regards
Gerard

      _case KEY_ICON_1:
        #ifdef MENU_LIST_MODE
          if(infoSettings.invert_yaxis == 1){
            storeCmd(YGCODE_DEC, item_move_len[item_move_len_i]);
          }
          else{
            storeCmd(YGCODE_INC, item_move_len[item_move_len_i]);
          }
        #else
          storeCmd(**YGCODE_UP**, item_move_len[item_move_len_i]);
        #endif
        break;
......
   
      case KEY_ICON_5:
        #ifdef MENU_LIST_MODE
          if(infoSettings.invert_yaxis == 1){
            storeCmd(YGCODE_INC, item_move_len[item_move_len_i]);
          }
          else{
            storeCmd(YGCODE_DEC, item_move_len[item_move_len_i]);
          }
        #else
          storeCmd(**YGCODE_UP**, item_move_len[item_move_len_i]);
        #endif
        break;_

guruathwal, 2020-01-03:
ok.. the Y move part is fixed now
 I unable to understand why custom g-codes are not working for you...
are you changing any part of the code ?

gneu42, 2020-01-03:
I don't know.
It works with Non list mode.

PLease find attached my configuration.h that does not work with me.

Regards
Gérard

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4019845/Configuration.zip)

guruathwal, 2020-01-03:
The config looks old. Did you updated your source code?
You can download the latest source with the new changes from here until the PR is merged.
https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware

gneu42, 2020-01-04:
Hi,
I downloaded your last version, and custom gcode now works in both mode. Thank you for the fix.
But some other problems are showing now. 
In List mode, the model icon does not show on the print screen (see video menu_list_mode.mp4)
In non list mode, model icons are showing eratically on the SD card screen and on the print screen (see video Non_menu_list_mode.mp4)
I include the zip file from the SD card with the bin file used, and the configuration.h in non menu_list_mode.

Regards 
Gerard
[SD-Card.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4021771/SD-Card.zip)

[Non Menu_list_mode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4021774/Non.Menu_list_mode.zip)

[Menu_List_mode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4021775/Menu_List_mode.zip)
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4021776/Configuration.zip)

guruathwal, 2020-01-04:
there has been some changes to the image preview part through this PR #272 by @chzj333.
@chzj333 might help on this.

Hukuma1, 2020-01-05:
@guruathwal The preview works fine in PR #272. I'm guessing this is before your latest changes?

Also I don't think we need the global icons on the print page (just as you removed them from the movement page). The info of temp is in the box to the right of the preview image. Not sure what happened, but they weren't there before and it looked proper? https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/157#issuecomment-569997200

guruathwal, 2020-01-31:
@gneu42  please close this issue if it has been solved.

Scope666, 2020-02-01:
On mine in List Mode, I see the 1st page but the "page down" icon does nothing, so you can't get to any of the other macros.

closed 2020-08-14 by bigtreetech