bigtreetech/BIGTREETECH-TouchScreenFirmware #275: [BUG] Does not compile with MENU_LIST_MODE enabled TFT35 v3

zech, 2020-01-01:
# Description

I'm trying to compile the latest firmware from the master branch myself with VSCode/PIO. As soon as I enable list mode it does not compile and throws a lot of  error messages like this:

```
TFT\src\User\Menu\Move.c:206:33: error: 'YGCODE_UP' undeclared (first use in this function); did you mean 'YGCODE_INC'?
```

It does seem like the renaming of variables was not done everywhere because it compiles fine with this option disabled.

### Steps to reproduce

1. Change `//#define MENU_LIST_MODE` to `#define MENU_LIST_MODE`
2. Compile

**Expected behavior**
Should compile without errors and warnings.

**Actual behavior**
Compiles with some errors as described above about variables not existing. And a ton of warnings about missing braces around initializers.

```
TFT\src\User\Menu\PreheatMenu.c:5:26: warning: missing braces around initializer [-Wmissing-braces]
 MENUITEMS preheatItems = {
```

### Additional Information
[Configuration_h.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4014302/Configuration_h.zip)

guruathwal, 2020-01-02:
Fixed here #279 

you can ignore the warnings, just compile again and they will disappear.

guruathwal, 2020-01-04:
@zech please close the issue

zech, 2020-01-04:
I can do when it's merged. Why isn't it so far?

guruathwal, 2020-01-04:
BTT is slow when it comes to merging. until then you can download the source with the fix from this repository: https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware

zech, 2020-01-04:
I see. The PR builds fine. Thanks. I'll close it then.

closed 2020-01-04 by zech