bigtreetech/BIGTREETECH-TouchScreenFirmware #276: babysteps unified menu

vMeph, 2020-01-01:
would it be possible add option for babysteps in unified menu

guruathwal, 2020-01-02:
it is already under ABL menu

vMeph, 2020-01-02:
@guruathwal yes i have seen that but i use UBL and that babysteps not doing nothing
i want the babysteps in realtime adjust while is printing the same way you do with the double click with enconder

guruathwal, 2020-01-02:
real-time babystepping is only possible through printer lcd. Babystepping through these TFTs is done through GCode, which have to wait in for the printer's gcode buffer to clear.

vMeph, 2020-01-02:
ya i understand, hopefully there will be away for someone to implement, right now thats the only function that is stoping me to go 100% into use the touch screen mode

guruathwal, 2020-01-02:
it will only be possible if Marlin firmware supports it. till then you can close the issue.

closed 2020-01-03 by vMeph